aredn/aredn #1371: High Contrast Impacts Tunnels and LAN DHCP

K5RA, 2024-08-23:
When using High Contrast theme (my theme of choice for now), the   network:port   parameters in the Tunnels config window and the   IP addresses   and   MAC addresses   in the LAN DHCP config window are larger than their cells. 

System is hAP ac lite with latest nightly, rp5b with latest bookworm, and latest Chromium browser.

Each cell should display full address.

aanon4, 2024-08-23:
Yes - the font is larger but there's less space as the dialog will not get bigger.

K5RA, 2024-08-23:
Work-around is to go back to Light theme.  Takes five mouse clicks to close screen, change theme, and go back to screen.  All IP and MAC characters are visible.  Three clicks to get back to High Contrast theme.

K5RA, 2024-08-23:
If for any reason I want to see the full cell contents, I can indeed click each cell individually, and pull cursor to right to expose the missing characters.  But I can use Light theme and see them all at once.   Example - technical discussions with associates, or understanding/maintaining my own system.  
In LAN DHCP window, each field appears to behave differently WRT clicking.  I will study it and get you a report. 
BTW - I am very grateful for the many hours of work you put in daily to bring this project along.

aanon4, 2024-08-23:
I'm going to drop the high contrast font base size by 1 pixel which might help in some circumstances.

aanon4, 2024-08-23:
https://github.com/aredn/aredn/pull/1376

closed 2024-08-24 by aanon4