aredn/aredn #1372: Admin pop-up doesn't pop - old raspbian chrome

jimsky7, 2024-08-23:
I'm on a raspberry Pi with a relatively old Raspbian and the default chromium browser. The icon for Admin (upper right corner of window) sometimes doesn't pop. And sometimes it pops but the popup is separated from the icon so when I move the cursor the popped menu disappears before I can click it. Simplest fix would be to pop at a position the cursor can move to without interruption.

aanon4, 2024-08-23:
Please update to a more modern browser.  I dont know how old the one you are using is, but if you are seeing these issue then it is so old as to be considered insecure and really shouldnt be used anymore.

jimsky7, 2024-08-23:
I updated to Firefox-ESR and same behavior. What I see is actually a tooltip. If I click the admin (little person) icon nothing happens. On my Mac in Firefox what happens is the "Login" menu pops open below the icon. But I need to access from Raspbian equally. The Firefox I am using on raspbian linux is "102.15.1esr (32-bit)"

aanon4, 2024-08-23:
Could I suggest moving to Firefox RR rather than ESR? Firefox 102 was originally released 2+ years ago. Firefox is currently on version 129.

jimsky7, 2024-08-23:
Attempting to debug >> The JS console shows this error "Password fields present on an insecure (http://) page. This is a security risk that allows user login credentials to be stolen." and then seems to suppress the password layer. I can look more at this in the JS debugger, but perhaps someone familiar with the code would know how I can get around this.

This message appears a minute later: "Partitioned cookie or storage access was provided to “https://worldmap.arednmesh.org/#12/37.7569/-122.45157” because it is loaded in the third-party context and dynamic state partitioning is enabled."

jimsky7, 2024-08-23:
Firefox RR not available on the distro I'm using, but I think the JS error is telling us this version of FF is preventing what it thinks is insecure behavior.

jimsky7, 2024-08-23:
But actually, clicking the login icon should open the 'Login" menu and it doesn't do that either.

jimsky7, 2024-08-24:
Working in Firefox JS console/debugger, the code that opens the 'Password' dialog can be fired off and the dialog appears and if I type my password, I can enter Admin mode with the node. So that's not an issue - the issue is that clicking the icon doesn't make the Login menu appear. I would suggest that requiring that the browser be less than 2 years old may not work in an Amateur Radio setting where people frequently repurpose old equipment and systems. having no access to Admin is going to be a problem that will knock out a significant number or participants. I will continue trying to help discover what is causing the click to not work. I see this issue has been marked 'not fix' and would like to help reverse that decision as it will make it impossible for me to admin a bunch of nodes...as well as other people now chiming in with this same problem. Perhaps the method being used to pop the login menu could be checked to see why it's not working with these browsers.

jimsky7, 2024-08-24:
The issue I had spotted with 'partitioned storage' turns out to be related to the node map, not to the admin process, so it can be ignored for the purposes of debugging the Login menu.

jimsky7, 2024-08-24:
Looking at the admin login menu code:

<div id="login-icon" class="popup-menu">
    <label title="Login as admin">
        <input type="checkbox">
        <div class="icon login"></div>
    </label>
    <div class="menu">
        <div>Login</div>
    </div>
</div>

...In the FIrefox Inspector the class 'menu' has 'display:none;' ... if I manually change that class to 'display:block;' in the inspector, the Login menu appears, I can click it, the Password dialog appears, and I can type the node's password and log in. BUT clicking the icon itself does not make the Login menu visible. I would ask why is the Input type Checkbox being used? I am not familiar with using that to make an object visible - unless attaching something like an 'onClick' to it and some JS code? Somehow in my browsers the Checkbox in this particular usage is not working? I also see there is SVG code for the icon, which is all fine, perhaps related to attaching that image to a 'checkbox' ? Again admitting I have not seen this specific type of coding for menus before, but wondering if an 'onClick' with JS to change the element's visibility would be more generally effective in more browsers. (And 'checkbox' has been around since 1995 so I can't write this off to the browser being too old - something else is failing here.)

aanon4, 2024-08-24:
So this is probably related to incomplete/buggy css implementations in older browsers. I say probably but it's an educated guess based on how things work and what you see. However, the most likely culprit is the CSS :has(...) construct which Firefox implemented a long time after it was added to the spec. Let me see how I can rework this.

aanon4, 2024-08-24:
https://github.com/aredn/aredn/pull/1384

aanon4, 2024-08-24:
Let's see if this fixes the login process.

This css construct is used extensively in the UI - but first let's see if this fixes this issue.

jimsky7, 2024-08-24:
Ah, yes I know about some of this more recent CSS and it makes sense it might be buggy on both Chromium and Firefox-EDR, and also on other older browsers like on Mac as reported recently.

I am not in SF today, however I have remote access to my nodes there so I can install a daily build via VNC. How should I determine when to do that install? When do the dailies get built?

Once I can reliably login as admin I can test the rest of the admin UI as well as install it on my Healdsburg nodes. I have only viewed little snatches of it so far and I find it easy to understand and to use.

I will have some comments about info that probably should not be in the public interface but instead should be visible only in admin mode. Such as the name of the Wi-Fi the node is connected to and its local IP address. Maybe also some other stuff.

- Sky

----
On Aug 24, 2024 at 1:31:33 PM, Tim Wilkinson wrote:
Let's see if this fixes the login process.This css construct is used extensively in the UI - but first let's see if this fixes this issue.—

aanon4, 2024-08-25:
This should take care of all the non-cosmetic stuff
https://github.com/aredn/aredn/pull/1386
A new build will be available in the morning. Would appreciate you testing.

closed 2024-08-26 by aanon4