aredn/aredn #1374: can't upgrade from nightly to anythign else

charlieh0tel, 2024-08-23:
### Describe the bug

my hap-ac3 was on 3.24.6.0.

i upgraded to nightly 20240818-4c57f60  to test the new UI.

i can no longer upgrade to the next nightly or back to 3.24.6.0.

I get:

Firmware CANNOT be updated
upgrade is not available

- [ ] Does this issue still occur on the latest version? (**Required**)

yes.

### Expected behavior
Able to upgade to next nightly or back to 3.24.6.0

### Screenshots

[image]

### Additional context

aanon4, 2024-08-23:
You need to use the new UI to upgrade or downgrade.

aanon4, 2024-08-23:
I will try to keep both working - https://github.com/aredn/aredn/pull/1375

aanon4, 2024-08-23:
Procedure:

To go back to 3.24.6.0 please use the firmware tools in the new UI. Refresh the download list and select the previous release.
Once downgraded you will need to restore the original firmware location in the advanced settings. You'll find a blank field towards the bottom of them in the Firmware section. To restore the default path press the "Set to Default" button then hit "Save".

charlieh0tel, 2024-08-23:
i flashed the latest nightly using the new UI.

thank you es 73.

closed 2024-08-23 by charlieh0tel