aredn/aredn #1378: Switching between UIs changes Tunnel Server IP Assignments

km6aow, 2024-08-23:
### Describe the bug
The IP address assigned to a user changes when switching between UIs.

- [ ] Does this issue still occur on the latest version? (**Required**) Yes - on version 20240823-c7d942a

### Expected behavior
I expect the tunnel IP addresses to remain as originally assigned.

### Screenshots
[image]
[image]
[image]

### Additional context
Note that the IP address 172.31.81.120 was assigned to KG6JHT under the old UI, and to WA6BXV under the new.

aanon4, 2024-08-23:
Please attach the support data. Thanks.

aanon4, 2024-08-23:
Also, it WA6BXV-NOVATO-HAP really in here three times?

km6aow, 2024-08-24:
I've asked him for the support data and will attach it when I get it. As for BXV, he said he made those additional entries to try to associate the correct IP address with the user after they changed. The table kept growning and he sees no way to delete obsolete entries.

km6aow, 2024-08-25:
[supportdata-AB6JR-Novato-hAP3-2024-8-24-7-35.tar.gz](https://github.com/user-attachments/files/16739008/supportdata-AB6JR-Novato-hAP3-2024-8-24-7-35.tar.gz)

aanon4, 2024-08-25:
Short version - the new UI is correct.
Longer version -
The new UI reads exactly what's in the config file for tunnels and display that information on the screen. The old UI reads the entries in order and assigned an IP address to each entry it displays based on the order it was read. That has been fine because the old UI always read and rewrote the entire list every time, so the order stayed the same. The new UI doesnt do that, instead using the information stored and only writing changes. The consequence of this is that, once the new UI has modified the tunnels, the tunnel order may no longer be the same, so the old UI will get it wrong.  There's not much to be done here because we have new features - like being able to delete tunnels - and the old UI just can't support this and get's confused.

closed 2024-08-30 by aanon4

aanon4, 2024-09-01:
Adding a warning: https://github.com/aredn/aredn/pull/1440