aredn/aredn #1379: Nightly 20240823-c7d942a (r23809-234f1a2efa) causes LHG 5nD to go deaf

kp4djt, 2024-08-24:
### Describe the bug 
Installed 20240823-c7d942a (r23809-234f1a2efa) a data link using two  Mikrotik LHG 5nD, the initial install on one end appeared to work just fine, installed on far end and both ends then went "deaf". I scanned for both radios but saw nothing.

- [ ] Does this issue still occur on the latest version? (**Required**)
- [ ] 20240823-c7d942a (r23809-234f1a2efa) is the latest nightly at this time

### Expected behavior
After loading of latest nightly I expected link to come back up, I waa able to reach the node on each end through the DtD
path, but not through the radio. A wifi scan revealed no RF signal 

### Screenshots
Have none. There was nothing to see. 

### Additional context
As indicated above, I did notice that the first upload of  20240823-c7d942a (r23809-234f1a2efa) to the initial LHG 5nD
showed a link connection, NOTE, both ends were runniing the previous rev dated 22 Aug 24. I had been doing updates as
new code came out.
Though this is a busy path it is easy to fix so I was using it as a test bed for the new UI code on a busy link.

aanon4, 2024-08-24:
Does rebooting both radios fix this?

kp4djt, 2024-08-24:
No it did not. The radios were talking just fine over the DtD port that is
how I got the stable loaded
on them, once 3.24.6 was on them they came right back up.

aanon4, 2024-08-24:
So after the update, the nodes did not reconnect as you expected, but you didnt try rebooting them before reverting them? As the networking and wifi code is identical in 3.24.6.0 and nightly (the UI is changing, nothing more) it would be a useful test.

kp4djt, 2024-08-24:
Yessire... I did. They just did not want to come back. As I indicated the
radio I was using to sniff the RF
was another LHG 5nD. It gave me  no issues

aanon4, 2024-08-25:
Not sure what to tell you - the radio code is the same on all the builds and on release.

kp4djt, 2024-08-25:
Well the latest is in there and they are cooking along just fine, so it is
resolved.

closed 2024-08-25 by aanon4