aredn/aredn #1383: pop-up images

nc8q, 2024-08-24:
**Is your feature request related to a problem? Please describe.**
In the New UI:
In the configuration pop-up images:
I cannot tell if there is a color difference in slider buttons.
I cannot tell what a button left or button right indicates...on/off,enable/disable

**Describe the solution you'd like**
A configuration page instead of an overlaid image pop-up.
I cannot do a clear print-screen because text in the grayed-out background also goes to the printer. 

**Describe alternatives you've considered**
How do I display/share tunnel credentials?

**Additional context**
[image]

K5RA, 2024-08-24:
New UI Documentation / Node Admin Guide / Add Tunnel / Wireguard Server gives instructions for getting server credentials that you can copy/paste into message to client.

aanon4, 2024-08-24:
@nc8q Not sure here .. are you saying the the grey / green contrast it difficult to see?

nc8q, 2024-08-25:
Hi, Tim:
The gray and dark-gray contrast is hard to see.
Without a comparison I cannot tell if the box was gray or darker-gray.
If the box is gray, what state does that signify?
If the box is darker-gray, what state does that signify?
[image]

aanon4, 2024-08-25:
I must be missing something still - sorry. When the switch is off, it is red. When the switch is on it is either gray (unconnected) or green (connected). There is no darker-gray.

nc8q, 2024-08-26:
I have deuteratopia.
I have a difficult time distinguishing among gray, red, and green,
It is not rare for me to be on a rooftop in bright sunlight configuring an AREDN node with a laptop.
I do well with radio buttons, drop-down text boxes, and hyperlinked text.
I fail when logic states are presented with only color options.
May I/we have a '1' or a '0' in the gray, red, or green rectangles?
73, Chuck

aanon4, 2024-08-27:
So what I'd prefer to do is develop a theme which is more appropriate for people with deuteranopia, in a similar way to the recent addition of the High Contract theme for people wanting larger text and less subtle grey levels.  The problem is that I dont understand what works and what doesn't; so I need to write a guide so someone who know what they're doing can make such a thing.

closed 2024-08-31 by aanon4