aredn/aredn #1385: NEW UI: Local Devices disappeared

mathisono, 2024-08-25:
NEW UI: Local Devices and & Local Services disappeared after amending legacy local service enters.

I pulled the first screen shot, pulled support data and then rebooted the node....

The second screen shot shows the node after a reboot, the new ui is broken, while the old ui sill show the amended entry.

  
[image]
[supportdata-W6BB-RFS-HUB4-2024-8-24-18-2.tar.gz](https://github.com/user-attachments/files/16738498/supportdata-W6BB-RFS-HUB4-2024-8-24-18-2.tar.gz)

--------
[image]

[image]

aanon4, 2024-08-25:
What were the specific changes? Can you back out the changes one by one and see if you can work out which specific change causes the problem? Thanks.

aanon4, 2024-08-25:
https://github.com/aredn/aredn/pull/1387

mathisono, 2024-08-25:
all i did was set an service icon/type

closed 2024-08-25 by aanon4