aredn/aredn #1390: UBNT Rocket M5XW doesn't show notes in New UI after upgrade N-20240824

k1ky, 2024-08-25:
### Describe the bug
A clear and concise description of what the bug is.
Node Notes aren't shown un new UI after upgrade from 3.24.5.0 to N-20240824.  

- [ ] Does this issue still occur on the latest version? (**Required**)
- [ ] Yes N-20240824

### Expected behavior
A clear and concise description of what you expected to happen.
Expect Notes to be shown.  They still exist on old UI screen.

### Screenshots
Please include a screenshot of your system information (if possible) if the specific system environment is relevant to the bug.
New UI: 
[image]
Old UI:
[image]

If applicable, add screenshots to help explain your problem.

### Additional context
I do have another RKM5XW that has upgraded to N-20240824 that shows description ok in New UI, so this may be an isolated occurrence.

k1ky, 2024-08-25:
Support Data File attached
[supportdata-KI4LMR-RKM5XW-103-LOVE-130TWR-000-2024-8-24-22-23.tar.gz](https://github.com/user-attachments/files/16738904/supportdata-KI4LMR-RKM5XW-103-LOVE-130TWR-000-2024-8-24-22-23.tar.gz)

aanon4, 2024-08-25:
Looks like it's because the description text has an = in it; which should be fine.

aanon4, 2024-08-25:
https://github.com/aredn/aredn/pull/1391

k1ky, 2024-08-25:
Yup - that was it.  I thought we were past the special characters limitation.  Thanks!

aanon4, 2024-08-25:
New code - new bugs :-) Fixed tomorrow.

k1ky, 2024-08-25:
Tried 20240825 and inserted "=" in notes in old UI and it now reflects the description in both old and new IU.

closed 2024-08-25 by k1ky