aredn/aredn #1395: Antenna elevation in what units?

jimsky7, 2024-08-25:
"Antenna elevation in degrees" - huh? I would measure antenna elevation in meters, not degrees. Azimutth in degrees, height (above ground) in meters are both understandable.

Orv, 2024-08-25:
Elevation is up or downtilt of the antenna

aanon4, 2024-08-25:
I believe that is the correct term - https://www.photopills.com/articles/understanding-azimuth-and-elevation

jimsky7, 2024-08-25:
OK then. In hilly country the traditional 'elevation' just seemed important to me, but with a topo map one could calculate that based on lat/lon.

jimsky7, 2024-08-26:
To K6RA, yes and that's why the UI wants them to be entered. All is well. I was confusing the civilian notion of 'elevation' as height above sealevel with the astronomical notion of 'elevation' as pointing upward, which is what this UI wants.

aanon4, 2024-08-26:
Assuming we're good here. Reopen if not.