aredn/aredn #1397: Enable POE doesn't work in new UI

Orv, 2024-08-25:
### Describe the bug
A brand new AC3.   Enabling and Committing POE out in the new UI doesn't work (shows enabled, but node on Port 5 doesn't power up.)

Does work in the old UI.

- [ ] Does this issue still occur on the latest version? (**Required**)
YEs
### Expected behavior
New UI should be able to control POE out on port 5

### Additional context
[supportdata-W6BI-TEST-AC3-2024-3-22-22-27.tar.gz](https://github.com/user-attachments/files/16742042/supportdata-W6BI-TEST-AC3-2024-3-22-22-27.tar.gz)

aanon4, 2024-08-25:
Could you try this experiment for me:
1. Disable it again using the old UI.
2. Enable it with the new UI (I assume it wont actually enable).
3. Reboot
4. See if it's now enabled.
Thanks

Orv, 2024-08-25:
That worked - booting with POE enabled in UI provided POE power out of port 5.

aanon4, 2024-08-25:
Thanks. I'll fix it (and usb power which is probably similarly bust)

aanon4, 2024-08-27:
https://github.com/aredn/aredn/pull/1412

Orv, 2024-08-27:
Now works as expected!

closed 2024-08-27 by Orv