aredn/aredn #1406: Base setup UI error after reboot. FW: 3.24.4.0-ath79 glinet_gl-ar300m16

bi7jta, 2024-08-26:
Hello, 
I got an error in this version firmware, 

`aredn-3.24.4.0-ath79-generic-glinet_gl-ar300m16-squashfs-sysupgrade.bin`

### Screenshots
[image]
 
See is a web browser (JS) issue

aanon4, 2024-08-26:
Could you provide an image of the main status page? Also, can you describe the exact steps you've taken to get to this point?

bi7jta, 2024-08-31:
> Could you provide an image of the main status page? Also, can you describe the exact steps you've taken to get to this point?

After reset value , it work again . I guess the wrong settings cause this issue.

[image]

closed 2024-08-31 by bi7jta