aredn/aredn #1407: Change "Antenna Elevation" label to read "Antenna Declination"

wu2s, 2024-08-26:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]
The label Elevation is easily confused with the Height above Ground. Declination may more accurately convey a sense of antenna tilt from horizontal.
**Describe the solution you'd like**
A clear and concise description of what you want to happen.
Change the one word label.
**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

K5RA, 2024-08-26:
If you check declination in Wiki or Google, you will find no definition bearing on the location of an antenna.  I am most familiar with declination meaning difference between magnetic north and true north, less so the hour-angle & declination coordinates describing locations of celestial bodies.  "Azimuth" and "elevation" are standard terms used to define the orientation of directional antennas, and those terms should probably continue to be used in AREDN.

wu2s, 2024-08-26:
One of the beauties of the English language is its ambiguity. If elevation is commonly understood among the radio community as meaning the up- or down-tilt of an antenna, then it should remain as it is.

wu2s, 2024-08-26:
No change needed

closed 2024-08-26 by wu2s