aredn/aredn #1414: Got git clone error in make process(docker mode)  error: RPC failed; curl 56 GnuTLS recv error (-9): Error decoding the received TLS packet.

bi7jta, 2024-08-27:
### Describe the bug
```
cp /opt/aredn/aredn/feeds.conf /opt/aredn/aredn/openwrt/feeds.conf
cat /opt/aredn/aredn/openwrt/feeds.conf.default >> /opt/aredn/aredn/openwrt/feeds.conf
cd /opt/aredn/aredn/openwrt; ./scripts/feeds uninstall -a
cd /opt/aredn/aredn/openwrt; ./scripts/feeds update -a
Updating feed 'arednpackages' from 'https://github.com/kn6plv/aredn_packages;openwrt-22.3.0-lua' ...
Cloning into './feeds/arednpackages'...
remote: Enumerating objects: 125, done.
remote: Counting objects: 100% (125/125), done.
remote: Compressing objects: 100% (98/98), done.
remote: Total 125 (delta 16), reused 107 (delta 14), pack-reused 0 (from 0)
Receiving objects: 100% (125/125), 5.56 MiB | 8.19 MiB/s, done.
Resolving deltas: 100% (16/16), done.
Updating feed 'packages' from 'https://git.openwrt.org/feed/packages.git^dba8a0102e5965cad58a871335002e9c964b6719' ...
Cloning into './feeds/packages'...
remote: Enumerating objects: 208799, done.
remote: Counting objects: 100% (208799/208799), done.
remote: Compressing objects: 100% (88035/88035), done.
error: RPC failed; curl 56 GnuTLS recv error (-9): Error decoding the received TLS packet.
fatal: the remote end hung up unexpectedly
fatal: early EOF
fatal: index-pack failed
failed.
Updating feed 'luci' from 'https://git.openwrt.org/project/luci.git^96ec0cd3ccfe954f13fd5a337efdd70374dde03f' ...
Cloning into './feeds/luci'...
remote: Enumerating objects: 39003, done.
remote: Counting objects: 100% (39003/39003), done.
remote: Compressing objects: 100% (6418/6418), done.
error: RPC failed; curl 56 GnuTLS recv error (-9): Error decoding the received TLS packet.
fatal: the remote end hung up unexpectedly
fatal: early EOF
fatal: index-pack failed
failed.
Updating feed 'routing' from 'https://git.openwrt.org/feed/routing.git^85028704f688a6768d3f10d5d3c10a799a121e0d' ...
Cloning into './feeds/routing'...
remote: Enumerating objects: 11448, done.
remote: Counting objects: 100% (11448/11448), done.
remote: Compressing objects: 100% (6954/6954), done.
remote: Total 11448 (delta 5269), reused 9083 (delta 3826), pack-reused 0
Receiving objects: 100% (11448/11448), 2.62 MiB | 2.40 MiB/s, done.
Resolving deltas: 100% (5269/5269), done.
Switched to a new branch '85028704f688a6768d3f10d5d3c10a799a121e0d'
/opt/aredn/aredn/openwrt
Updating feed 'telephony' from 'https://git.openwrt.org/feed/telephony.git^1d2031a5c82816483c51bca15649e2957fbe2bc2' ...
Cloning into './feeds/telephony'...
remote: Enumerating objects: 11912, done.
remote: Counting objects: 100% (11912/11912), done.
remote: Compressing objects: 100% (7283/7283), done.
remote: Total 11912 (delta 6705), reused 8140 (delta 4394), pack-reused 0
Receiving objects: 100% (11912/11912), 2.39 MiB | 2.24 MiB/s, done.
Resolving deltas: 100% (6705/6705), done.
Switched to a new branch '1d2031a5c82816483c51bca15649e2957fbe2bc2'
/opt/aredn/aredn/openwrt
Create index file './feeds/arednpackages.index' 
Checking 'working-make'... ok.
Checking 'case-sensitive-fs'... ok.
Checking 'proper-umask'... ok.
Checking 'gcc'... ok.
Checking 'working-gcc'... ok.
Checking 'g++'... ok.
Checking 'working-g++'... ok.
Checking 'ncurses'... ok.
Checking 'perl-data-dumper'... ok.
Checking 'perl-findbin'... ok.
Checking 'perl-file-copy'... ok.
Checking 'perl-file-compare'... ok.
Checking 'perl-thread-queue'... ok.
Checking 'tar'... ok.
Checking 'find'... ok.
Checking 'bash'... ok.
Checking 'xargs'... ok.
Checking 'patch'... ok.
Checking 'diff'... ok.
Checking 'cp'... ok.
Checking 'seq'... ok.
Checking 'awk'... ok.
Checking 'grep'... ok.
Checking 'egrep'... ok.
Checking 'getopt'... ok.
Checking 'realpath'... ok.
Checking 'stat'... ok.
Checking 'unzip'... ok.
Checking 'bzip2'... ok.
Checking 'wget'... ok.
Checking 'install'... ok.
Checking 'perl'... ok.
Checking 'python2-cleanup'... ok.
Checking 'python'... ok.
Checking 'python3'... ok.
Checking 'python3-distutils'... ok.
Checking 'git'... ok.
Checking 'file'... ok.
Checking 'rsync'... ok.
Checking 'which'... ok.
Checking 'ldconfig-stub'... ok.
Collecting package info: done
Collecting target info: done
Create index file './feeds/packages.index' 
find: 'feeds/packages': No such file or directory
Collecting package info: done
find: 'feeds/packages': No such file or directory
Collecting target info: done
Create index file './feeds/luci.index' 
find: 'feeds/luci': No such file or directory
Collecting package info: done
find: 'feeds/luci': No such file or directory
Collecting target info: done
Create index file './feeds/routing.index' 
Collecting package info: done
Collecting target info: done
Create index file './feeds/telephony.index' 
ERROR: please fix feeds/telephony/net/freeswitch/Makefile - see logs/feeds/telephony/net/freeswitch/dump.txt for details
ERROR: please fix feeds/telephony/net/kamailio/Makefile - see logs/feeds/telephony/net/kamailio/dump.txt for details
Collecting package info: done
Collecting target info: done
make: *** [Makefile:80: .stamp-feeds-updated] Error 1
```
[image]

[image]

aanon4, 2024-08-27:
Sorry, but I cannot even begin to guess what you're trying to do here, so there's not much help I can provide.

bi7jta, 2024-08-31:
Thank you for reply , I have fixed , the reason is my network too slow.

> Sorry, but I cannot even begin to guess what you're trying to do here, so there's not much help I can provide.

closed 2024-08-31 by bi7jta