aredn/aredn #1419: NLQ still > 100%

Orv, 2024-08-28:
### Describe the bug
NLQ still shows > 100% on some remote nodes (with tunnel weight >1 )

- [ ] Does this issue still occur on the latest version? (**Required**)
Yes - both ends current
### Expected behavior
A clear and concise description of what you expected to happen.
Should never be > 100%
### Screenshots

[image]

[supportdata-W6BI-SHACK-NODE-2024-8-28-13-52.tar.gz](https://github.com/user-attachments/files/16788825/supportdata-W6BI-SHACK-NODE-2024-8-28-13-52.tar.gz)

aanon4, 2024-08-29:
https://github.com/aredn/aredn/pull/1420

aanon4, 2024-08-30:
Fixed?

Orv, 2024-08-30:
Works as expected.

closed 2024-08-30 by Orv