aredn/aredn #1423: New UI allows forced DHCP option without tag

pmilazzo, 2024-08-29:
### Describe the bug
In the LAN DHCP Advanced Options table, the New UI allows the "Always" checkbox to be checked even though no tag has been selected to restrict the applicability of that option. While technically a valid&mdash;though unlikely&mdash;configuration choice, this combination cannot be expressed in the current version of OpenWRT UCI and should thus be suppressed in the UI.

- [x] Does this issue still occur on the latest version? (**Required**)

### Expected behavior
- Whenever the tag value is [all], the "Always" checkbox should be cleared and disabled.
- When the tag value is changed from [all] to a specific tag, the "Always" checkbox should be enabled but remain in the cleared state.
- When the configuration is read from a file, an option that has no tag but specifies "force" (presumably because it was written with an earlier version of the firmware) should be overwritten with "onrequest".

aanon4, 2024-08-30:
At the moment there is none of you fancy option/tag validation work - I didnt port it when I was getting this feature implemented.

aanon4, 2024-11-26:
This should include the 'always' flag fix.
https://github.com/aredn/aredn/pull/1718

aanon4, 2025-02-04:
Please verify this is fixed.

closed 2025-02-05 by aanon4