aredn/aredn #1424: Upgrading a Nanobeam from Nightly causes error

wu2s, 2024-08-29:
### Describe the bug
A clear and concise description of what the bug is.
Upgrading a Nanobeam (old unit) from Nightly Build 29249826 to 20240829 causes an Unknown error. This does not happen on other units.

### Expected behavior
A clear and concise description of what you expected to happen.
Expected upgrade to install successfully.

### Screenshots
Please include a screenshot of your system information (if possible) if the specific system environment is relevant to the bug.

[image]
[Nanobeam Error.pdf](https://github.com/user-attachments/files/16805923/Nanobeam.Error.pdf)

### Additional context
Add any other context about the problem here.

k1ky, 2024-08-29:
Why does the installed firmware show Nanostation Loco-M-XW?  What is the exact model of this unit?

wu2s, 2024-08-29:
The unit is a Nanobeam M2 13. A very old flying saucer looking unit about 7 inches in diameter. I went through a series of upgrades from 3.18.9.0 to the NB 20240826 successfully. The Loco-m-xw firmware is what got selected at some point in the upgrade series, and it worked just fine.

aanon4, 2024-08-30:
Could you see if you can update by downloading the firmware from the website and then using the upload option?

aanon4, 2024-08-30:
Also, if this then get the same error, could you send me the support data immediately after you see it? Thanks.

wu2s, 2024-08-31:
Attempted to load another nightly build as you requested. The upload stopped without completing successfully. Unit appears to be hosed. I plan to try starting over with a recent stable factory build to recover the unit.
[Nanobeam Error Aug 30.pdf](https://github.com/user-attachments/files/16825077/Nanobeam.Error.Aug.30.pdf)

ab7pa, 2024-08-31:
I have a 10 year old Ubiquiti NanoBridge M2 that upgrades successfully to the latest nightly builds, but there is an AREDN firmware image specifically for the nanobridge.  It does not rely on a "loco" image.
[image]

aanon4, 2024-08-31:
Stopping halfway though downloading the image suggests its running out of RAM to store it.  If you can disconnect it from the network first that might help.

ab7pa, 2024-08-31:
[I wonder if this might work](https://docs.arednmesh.org/en/latest/arednHow-toGuides/firmware_tips.html#tips-for-upgrading-firmware):

> **Try to Sideload Firmware**
> The Sideload Firmware option is described in the Node Admin guide. This involves using a file copy utility on your computer to copy the firmware file to a specific directory and filename on your node. Once the new firmware file is available on the node, you can click the Update button to start the install process.

wu2s, 2024-09-01:
Tried again with this unit disconnected from the network. Attempted to load firmware directly from computer to node without success.
Probably too little memory left as this screenshot shows. Looks like this model has hit the end of support life.
[WU2S-NBE-A Memory.pdf](https://github.com/user-attachments/files/16829608/WU2S-NBE-A.Memory.pdf)

aanon4, 2024-09-01:
Is it possible to upgrade the node, but do it from the old UI instead?

aanon4, 2024-12-04:
Any update on this problem?

closed 2025-07-02 by aanon4