aredn/aredn #1425: New UI cannot handle unlisted DHCP option numbers

pmilazzo, 2024-08-29:
### Describe the bug
The New UI can neither create nor display DHCP options with a code (tag octet) that is not in the list of known option types. Only known options can be chosen in the &lt;select&gt; element, and previously configured options with an unknown code are displayed as an empty field.

The known options support many but not all DHCP use cases. There are many more defined options than the ones known to dnsmasq, and it is common for manufacturers to require private DHCP codes for certain devices.

- [X] Does this issue still occur on the latest version? (**Required**)

### Expected behavior
The direct way to support unknown options would be to use a &lt;datalist&gt; element in place of the &lt;select&gt; element. This element is now supported in most major browsers on most platforms, but not all.

The old UI used a special "[other option number]" choice in the &lt;select&gt; element:
[image]

...which, when selected, would swap out the &lt;select&gt; element for a numeric input field:
[image]

When the option was added to the table, the formerly unknown option code would be interpolated into the &lt;select&gt; element:
[image]

Sadly, when this configuration is displayed in the new UI, the option code remains unknown:
[image]

aanon4, 2024-08-30:
So I think the way to go here is to use a feature called 'datalist' which provides autocomplete to input fields (originally implemented in 2011 ... I discovered them today). This should allow someone to enter a number or select from a dropdown list of known options.

aanon4, 2024-08-30:
https://github.com/aredn/aredn/pull/1430

pmilazzo, 2024-08-30:
Great idea! The &lt;datalist&gt; approach seems to work well (I tried it in Chrome on Linux and Edge and Firefox on Windows).

One trick it helps to know: the input field will only display the list when empty. So, to switch to a different choice you first have to clear the input field and then re-open the list.

aanon4, 2024-08-30:
On Safari (and it's probably a browser dependent thing) the list will re-open as soon as you start typing so long as there's a partial match.

aanon4, 2024-08-31:
@pmilazzo close?

pmilazzo, 2024-08-31:
Yes, close it. I'm happy.

closed 2024-08-31 by aanon4