aredn/aredn #1431: NEW UI: unable to disable a service link

mathisono, 2024-08-30:
I the old UI i could simply unchecked the box or the advertised service, and the service link would be disabled from advertisement. The new UI lacks this...

[image]

[image]

mathisono, 2024-08-30:
The expected behavior is to allow the user to disable a listed service from being advertised or shown (IE the serves is down and i dont what it advertised or shown)

aanon4, 2024-08-30:
That's not what that checkbox does. It changes the service from being something you can click on to something you can't. It's still advertised.

K5RA, 2024-08-30:
The Help on the Local Services config page does not provide info about the various cells and buttons.  Neither does the brief description of Local Services in the Node Status Display section of the new UI docs.  However, in the new UI docs, Node Admin Guide/Local Service section is excellent.  Viewing, Editing, and Deleting Services may have what you need.

K5RA, 2024-08-30:
@aanon4 
Thanks for the explanation of the checkbox.  Two days ago there was an advertised service on our mesh that could not be "accessed."  I had never seen that before.  It's gone now, so guess owner clicked Del.

aanon4, 2024-08-30:
A node will check that a service is reachable. If this check fails (it makes a few attempts) the service stops being advertised automatically. Once it returns, it will be automatically re-advertised.  We did this, as you observer, people often advertise broken services and its better we don't do that.

@mathisono we can turn this into a feature request if you'd like; but this was never an option in the old UI.

mathisono, 2024-08-31:
It would be useful to not have to delete an "advertised service" to not advertise it!  

How about a check box with the same option as with LAN address reservations "do not propagate" 

I don't need the node to check the advertisements ive posted for the MTX proxies.  How about a check box to enable the Automatic check...?

aanon4, 2024-08-31:
So because we dont manage any enable/disable state for services right now, supporting this would break the old UI; so this will have to wait until the old UI is deprecated.

jjkroell, 2024-11-21:
Add the service as ‘Simple Text’