aredn/aredn #1434: Unable to set static WAN ip

kodiakastu, 2024-08-30:
### Describe the bug
I just provisioned a VM with the today's nightly (20240830). First time trying to setup a VM. When I set a static IP in the new UI, the IP turns to null ( see 1st screenshot).
 
Tried to setup in OLD UI, but when you change the wan type, you're missing half of the fields.

### Expected behavior
The static IP should be saved.

### Screenshots
[image]
[image]

### Additional context
I have 3 adapters set in proxmox and they're configured like this:
[image]

If I left WAN on DHCP it works as intended.

aanon4, 2024-08-30:
Please attach a screenshot of the open network setup dialog. Also please add the support data. Thanks.

kodiakastu, 2024-08-31:
Like this ?
[image]
[supportdata-VA2LRM-LPLN-Q01-VM00-2024-8-30-19-54.tar.gz](https://github.com/user-attachments/files/16822822/supportdata-VA2LRM-LPLN-Q01-VM00-2024-8-30-19-54.tar.gz)

aanon4, 2024-08-31:
Thanks. Did you enter values into those boxes?

kodiakastu, 2024-08-31:
Yes. Before I click done:
[image]
Once I click done:
[image]
Once I click commit:
[image]

aanon4, 2024-08-31:
https://github.com/aredn/aredn/pull/1435

aanon4, 2024-08-31:
Should be fixed tomorrow

kodiakastu, 2024-08-31:
That was quick! When does the nightly builds ? I'll update as soon as I can to test.

aanon4, 2024-08-31:
They're usually available by 13:00 UTC each day

aanon4, 2024-08-31:
Let me know if today's build works for you.

kodiakastu, 2024-08-31:
Works as intended on my current vm. Thanks again!

closed 2024-08-31 by kodiakastu