aredn/aredn #1441: Switching themes - regression

Orv, 2024-09-01:
### Describe the bug
A clear and concise description of what the bug is.
In the new UI, the ability to switch themes requires you to be logged in.  This is a regression from the behavior of the old UI.   The ability to switch themes on any node regardless of whether your logged in should be preserved.
- [ ] Does this issue still occur on the latest version? (**Required**)
Yes

aanon4, 2024-09-01:
Absolutely not. The ability of a random person to change a theme on my node without permission was just broken.

Orv, 2024-09-01:
Huh - I viewed it as a service to other hams who might have difficulty looking at your node's settings.

aanon4, 2024-09-01:
If it were just that - viewing a node with a theme of your choosing - I'd be for it, but it's not. It changes what everyone sees, and not what the operator has chosen.

That said, I've been thinking about a feature to allow you to use your localnode theme when looking at other nodes. I think that would both be a better approach and respect the ownership of an operator's device. What do you think?

ab7pa, 2024-09-01:
I agree with Tim.  I've never liked that a non-node-owner could change the theme and impact how everyone else viewed the UI.  I suggest that we keep Tim's approach until we figure out a way to allow the UI theme to be changed on a browser-by-browser basis -- with multiple people viewing the UI using different themes at the same time.

Orv, 2024-09-01:
I'm fine with that approach.

aanon4, 2024-09-01:
https://github.com/aredn/aredn/issues/1442

closed 2024-09-01 by aanon4