aredn/aredn #1442: Bring along your own theme when viewing other nodes

aanon4, 2024-09-01:
With the old UI anyone could change the theme on your node. This could be useful for the random viewer, but the node was left in a state not selected by the operator. The new UI requires login to change the theme.

Instead, this feature allows you to bring your localnode theme along with you, and use that to view other nodes. It does not change the node's and the view is personal only to you.

aanon4, 2024-09-01:
@Orv @ab7pa Opinions: It would be very easy to make this behavior the normal behavior as it seems generally better. But I could also make it off by default, or let nodes opt-out of it. I'm loathed to put in many options people mostly wont use or want; so wanted your thoughts on this?

Orv, 2024-09-01:
I'm fine with it defaulting off.   And lately I've been worried about "feature-itis"   A ham friend of mine is going to submit a couple of issues requesting that some 'basic' setups be the startup defaults.  They sounded reasonable to me.   We'll see....

K5RA, 2024-09-01:
How is this going to work with the seven themes in various color combinations in the old UI, and the four (light, highcontrast, dark, and colorblind) in new UI??  Will user bring his old or new UI with him to the remote node with whatever is the theme of user node?  Does the remote node have to have new UI installed for it to work?  What if user with old node goes to remote node (future) that has no old UI available or vice versa?

aanon4, 2024-09-01:
This is only about the new UI. No plans for any of this in the old UI.

K5RA, 2024-09-01:
The "only about the new UI" is the part about taking your theme with you to the remote node?  
The ability to go to a remote node, change the theme, and leave it changed is an old UI thing that will remain, or be deprecated?
I have never used other than AREDN theme in old UI, so this is only for my understanding.

aanon4, 2024-09-02:
https://github.com/aredn/aredn/pull/1443

closed 2024-09-02 by aanon4