aredn/aredn #1445: New UI Supernode tunnel builds incorrectly

bpaplow, 2024-09-02:
Using the new UI on a fresh setup does not build the wireguard tunnels correctly. It uses the default 5526 and not the 6526 that supernodes use.

happens in nightly to at least 20240831-638bc9a

When a tunnel is built in the old UI, the new one shows it correctly.
When a tunnel was built before the new UI it imports correctly.

steps to repeat, 
1. Install node with 3.24.6.0 clean. enable supernode mode per instructions.
https://docs.arednmesh.org/en/latest/arednHow-toGuides/supernodes.html
2. Upgrade to nightly
3. From new UI start new WG tunnel server.

New UI shows 5526 as the port and the config /etc/config/network WGC0 show port 5526 old UI show 6526 (this is not correct)

From Old UI start new WG tunnel server

New UI shows 6526 as the port and the config /etc/config/network WGC0 show port 6526 old UI show 6526 (this is correct)

aanon4, 2024-09-02:
https://github.com/aredn/aredn/pull/1446

aanon4, 2024-09-05:
Did this resolve the problem?

bpaplow, 2024-09-05:
The initial problem is solved in that the first tunnel builds with 6526 now. The new issue is it only build 6526 for any additional tunnels creating duplicate tunnels servers on the same port and network where only the key and name is changed.

aanon4, 2024-09-05:
Could you post screen shots? Thanks.

bpaplow, 2024-09-06:
Reset the config instead of just removing the tunnels and now its working as expected.

closed 2024-09-06 by aanon4