aredn/aredn #1447: Portable Theme Not Working?

K5RA, 2024-09-02:
System: hAP ac lite with 20240902 nightly, R-Pi 5B with latest bookworm, Chromium 126 browser.

Downloaded latest firmware.  Opened Basics window, turned on Portable Theme. Done. Commit.
Hopped to a local node.  Was able to carry my local theme to that node.  High Contrast, then Color Blind, then High Contrast. 
Hopped to local neighbor.  Node shows running 20240831.  Showed not logged in as expected.  Theme did not transfer.  Tried several times.  Also, my theme did not transfer to local neighbors running 3.24.6.0.

Maybe need to clarify "any node" in Basic window Portable Theme set up.

aanon4, 2024-09-02:
Unless those nodes have also been upgrade, you are not going to see this on them.

aanon4, 2024-09-02:
Nodes must be running 20240902-4ab2629 or later.

K5RA, 2024-09-02:
My NanoLoco M2 XM is running 20240823.  It is DTD to my hAP ac lite running 20240902-4ab....  The NanoLoco, my Local Node, has repeatedly taken themes from the hAP which has run both 20240902 nightlys.   I am not sure about the "must be running."  
Also, when I click the AREDN logo on the NanoLoco, it browses to  localnode.local.mesh/   unlike the other nodes that browse to   localnode.local.mesh:8080.

closed 2024-09-02 by K5RA