aredn/aredn #1450: Knightly Build 20240902-2ac72eb CSS formatting

smfichera, 2024-09-02:
### Describe the bug
I updated my hAP lite to Knightly Build 20240902-2ac72eb and the css formatting is not displaying correctly.  Looking at the html source I see references to localnode.local.mesh in the head.  This particular node is a dtd.  The node that resolves to localnode is not yet on the knighly build firmware because it is a production tunnel server that I only run final releases on so there is no place for the browser to pull the .css file.  I tried multiple themes with the same result.

- [X] Does this issue still occur on the latest version? (**Required**)

### Expected behavior
Formatting of the page has a problem

### Screenshots
[image]

### Additional context
Rebooted the node in addition to the reboot during and upgrade.  Also deleted browser cache.

aanon4, 2024-09-02:
https://github.com/aredn/aredn/pull/1449

aanon4, 2024-09-02:
Please update.

closed 2024-09-02 by aanon4