aredn/aredn #1452: With WiFi Scan, 1 node is showing twice

DaleJag, 2024-09-02:
I am running the latest nightly version today 9-2-24 @ 11:30 am.  20240902-4ab2629
The node is set for 10 Mhz on Ch 167.  There are no other nodes nearby on 10 Mhz.
There are other nodes in the area that are on 20 Mhz.

On 8-30-24 running an older nightly (about 20240828)
When I run a WiFi Scan, I get 2 readings from the same node.
One has an SSID of AREDN-10-v3    Which should be correct because it is running at 10 Mhz.
The other SSID is AREDN-20-v3   Which should never come up when running 10 Mhz.
[image]

Then today,  9-2-24 @ 11:30 am.  20240902-4ab2629 I did an upgrade to the nightly build.
This is what I got for a WiFi Scan.
[image]

My questions are:
1.   Why is this node showing a 20 Mhz listing?  First pic.
2.    Why am I getting 2 different MAC addresses for the same node that is running SSID: AREDN-10-v3?    Before the Nightly update the MAC address was:  2A:78:32:85:13:44         Now, after the nightly update the MAC address is:  32:6D:C3:37:95:99

Mesh:  10.233.210.51
Lan:  10.78.145.153
[supportdata-WA6MZW-LGH5AC-CCCove2DHS-202409021922.tar.gz.zip](https://github.com/user-attachments/files/16840828/supportdata-WA6MZW-LGH5AC-CCCove2DHS-202409021922.tar.gz.zip)
For Password E Mail me at:  WA6MZW@MailByChoice.com

I have seen this problem on different nodes in the past, but never reported it to you.  It did go away months later.  In the mean time, this node does not receive a good signal from a distant node.  The node has been rebooted a multiple of times and even had a complete power disconnect.  Problem continues.

### Describe the bug
A clear and concise description of what the bug is.

- [ ] Does this issue still occur on the latest version? (**Required**)

### Expected behavior
A clear and concise description of what you expected to happen.

### Screenshots
Please include a screenshot of your system information (if possible) if the specific system environment is relevant to the bug.

If applicable, add screenshots to help explain your problem.

### Additional context
Add any other context about the problem here.

aanon4, 2024-09-23:
I would be good to see what the new UI wifi scanner shows.
As for not showing 20 MHz SSIDs ... it all depends on the wifi chipset and low level driver. Some provide everything and some do not. We dont control that.

ae6xe, 2024-09-23:
The "iw dev wlan0 scan" output in the support data file:

BSS 0e:3d:d1:28:f4:20(on wlan0)
	last seen: 2962.947s [boottime]
	TSF: 800446716199 usec (9d, 06:20:46)
	freq: 5835
	beacon interval: 100 TUs
	capability: IBSS (0x0002)
	signal: -75.00 dBm
	last seen: 0 ms ago
	Information elements from Probe Response frame:
	SSID: AREDN-20-v3
BSS 32:6d:c3:37:95:99(on wlan0) -- **joined**
	TSF: 0 usec (0d, 00:00:00)
	freq: 5835
	beacon interval: 100 TUs
	capability: IBSS (0x0002)
	signal: 0.00 dBm
	last seen: 2834900 ms ago
	Information elements from Probe Response frame:
	SSID: AREDN-10-v3

Two comments:

1)  the scan reports "My adhoc network" for this "AREDN-20-v3" SSID, but we are 'joined' to another network.   It should probably say "foreign adhoc network" when it sees this.

2) This bogus entry was last seen 50 min after boot -- not sure what the uptime is when this was found.  Possibly the change from 20 to 10Mhz channel was recently done without rebooting linux and ath10k continues to cache info.  I see settings on cache timeout, reported to be hours.   copilot reports the interface has to be brought down, then back up to clear the BSS cached info.

Joe

DaleJag, 2024-09-28:
After changing this to 10Mhz, I did a reboot.  Same 2 readings came up.
Then I did a complete power cycle down for 10 seconds.  Same 2 readings came up.
This is the same node showing at 10Mhz & 20Mhz with 2 different MAC addresses at the same frequency and Node Name.
[image]