aredn/aredn #1455: NTP auto-sync not working on Basebox 5, QRT 5, mANT 19S

AI7NC, 2024-09-03:
### Describe the bug
On Basebox5, QRT 5, and mANT 19S hardware devices running 3.24.6.0 (also was issue on older versions), it has been found that their time clocks get out of sync by 10s of minutes in some cases hours. We do not see this issue with any offline hAPs, either lite, or AC2 they sync with our time server and display the correct time/date.

Time is nearly always out of sync by a non-hour integer rounding to a few minutes so it doesn't appear to be a time zone issue. Also changing time zone to UTC does not resolve the issue. 

Logs do not show any errors but do show sync attempts to the correct NTP time server pbx-lccore.local.mesh they look the same as nodes that are maintaining time well. Examples elsewhere in the mesh are pointed to different mesh NTP servers. Nodes other than these hardware types have no issue syncing with pbx-lccore. We have a mix of daily and hourly sync with these hardware types, neither work correctly. 

Here's an example of a node not syncing correctly, support is also attached. Time should be 23:03 but is 22:33 on this node, not all are a perfect 30 min offset.
[image]

Config:
[image]

Checking that the web settings have modified the openwrt system config for ntp
[image]

Testing NTP service to the lccore-pbx rom this node without changing local clock to verify contact should work. 
[image]

Manually running 'ntpd -q -p lccore-pbx.local.mesh' via SSH sets the correct time on the device.
Here's before the manual sync but after being online for days with ntp set to sync hourly with lccore-pbx.
[image]

Here's after the manual sync, time and date is now correct.
[image]

We've found that the manual sync works with multiple NTP servers we're trying to auto-sync to and all three hardware types. Our suspicion is there's a misconfiguration somewhere in the firmware for these hardware types, possibly at the openwrt level that is causing the NTP sync to not work correctly. Manual update successes indicates it is not a AREDN configuration, network, or NTP server issue. Manual sync doesn't generate any syslog entries in the support file unless the NTP server is unreachable.

### Expected behavior
Hourly sync with NTP keeping the clock within a minute or two of accurate. 

### Additional context
Florence group noted this has been an issue for years for them across many firmware iterations.

[ntpworking-AI7NC-M5-East-Springfield-202409030927.tar.gz](https://github.com/user-attachments/files/16852153/ntpworking-AI7NC-M5-East-Springfield-202409030927.tar.gz)
[ntpNOTworking-W7PXL-LCCORE-EAST-202409030851.tar.gz](https://github.com/user-attachments/files/16852154/ntpNOTworking-W7PXL-LCCORE-EAST-202409030851.tar.gz)
[ntpNOTworking-KI7TRH-PHRB-Sector-202409022351.tar.gz](https://github.com/user-attachments/files/16852198/ntpNOTworking-KI7TRH-PHRB-Sector-202409022351.tar.gz)
[ntpNOTworking-WA7ABU-M5-QRT-NEWBERG-202409030655.tar.gz](https://github.com/user-attachments/files/16852671/ntpNOTworking-WA7ABU-M5-QRT-NEWBERG-202409030655.tar.gz)

aanon4, 2024-09-03:
Questions. So in your example when you sync by hand, how long does it take before the clock is significantly different again? And how fast does it drift away from the correct time?  Also, do you happen to have the same hardware which doesnt show this issue?

AI7NC, 2024-09-03:
The node I manually synced and provided screenshots of in the post this
morning is already off by a lot.

[image: image.png]

73,

Nathan Campbell, AI7NC

aanon4, 2024-09-03:
Looking through the log of one of these nodes, we see the time being updated:
```
Sat Aug 31 16:45:55 2024 user.notice update-time: Update clock from lccore-pbx.local.mesh
Sat Aug 31 17:05:59 2024 daemon.info olsrd[1862]: Received netlink error code Invalid argument (-22)
```
and
```
Mon Sep  2 07:10:05 2024 user.notice update-time: Update clock from lccore-pbx.local.mesh
Mon Sep  2 08:10:09 2024 daemon.err manager.lua[1636]: ping: bad address 'downloads.arednmesh.org'
```
So it looks like the periodic updating of the internal clock is at least happening. This is on `KI7TRH-PHRB-Sector` which is set to update the time daily.  How does hourly effect this node?

I went to look at the data for Thurston (you have pictures above) but it wasn't attached. It would be useful to see these logs.

AI7NC, 2024-09-03:
We see the updates in all the logs but the clocks are far off. Here's three more support file pulls.

[supportdata-W7PXL-LCCORE-EAST-202409031425.tar.gz](https://github.com/user-attachments/files/16856526/supportdata-W7PXL-LCCORE-EAST-202409031425.tar.gz)
[supportdata-KI7TRH-PHRB-Thurston-202409031406.tar.gz](https://github.com/user-attachments/files/16856528/supportdata-KI7TRH-PHRB-Thurston-202409031406.tar.gz)
[supportdata-KI7TRH-PHRB-Sector-202409031239.tar.gz](https://github.com/user-attachments/files/16856529/supportdata-KI7TRH-PHRB-Sector-202409031239.tar.gz)

aanon4, 2024-09-03:
Oh I get that, I just wanted to logs to go with the screenshot. Thanks.

aanon4, 2024-09-03:
Experiment:
Pick one of the broken nodes; doesnt matter which one, but one with > 32MB of memory and I suppose the worst might be also make good test.
Log in and edit `/etc/config.mesh/system`. Find the lines:
```
config timeserver 'ntp'
        list server '<ntp_server>'
        option enable_server '0'
        option enabled '0'
```
And change `option enabled '0'` to be `option enabled '1'`
Run `/usr/local/bin/node-setup`
Run `/etc/init.d/sysntp start`
Run `ps | grep ntpd` to make sure ntpd is now running.
Let's see if its a poorly behaving hardware clock that's the cluprit (and so running ntpd 'properly' should help) or something else is going on.

aanon4, 2024-09-03:
https://github.com/aredn/aredn/pull/1457

AI7NC, 2024-09-04:
I have tested this on the W7PXL basebox 5 node. It seems to have worked as the time was fixed after configuring and starting the service.

[image]

AI7NC, 2024-09-04:
It seems to be sliding behind rather quickly.

[image]

aanon4, 2024-09-04:
So in 40 minutes (realtime) the node thinks the time changed by 25 minutes .. and the ntpd daemon is still running?

AI7NC, 2024-09-04:
The process is still running: 

[image]

[image]

aanon4, 2024-09-04:
Let's see how it changes over time.  Looks like the daemon pings its peer every 11 minutes so, overtime, I assume it brings the system time and incremental time in check.

aanon4, 2024-09-04:
And assuming it continues to drift, it might be interesting to see some debug information. Stop the daemon using ```/etc/init.d/sysntp stop```
then run it by hand at the console
```/usr/sbin/ntpd -n -N -ddd -S /usr/sbin/ntpd-hotplug -p us.pool.ntp.org```
replacing the peer on the end with whichever one you're using

AI7NC, 2024-09-04:
It is currently 31 minutes behind.

[image: image.png]

73,

Nathan Campbell, AI7NC

aanon4, 2024-09-04:
So:
In 40 minutes it lost 15 minutes
In 180 minutes it lost 31 minutes
Let's check on it tomorrow. Note that you probably have the standard update-clock script still running every day/hour which you might want to disable so it doesnt mess with this test (you can find it in either /etc/cron.hourly or /etc/cron.daily)

AI7NC, 2024-09-04:
I'm guessing it synced at some point as we're 17 minutes behind this morning. 

[image]

aanon4, 2024-09-04:
Did you disable the cron time update?

AI7NC, 2024-09-04:
I pointed the standard NTP server to something that shouldn't return a time. Will that not work?

aanon4, 2024-09-04:
For testing purposes just move the "update-clock" file somewhere else .. into /etc will do just do.

aanon4, 2024-09-04:
And unless you are running the ntpd server by hand, you do need to give it a valid time source in the standard configuration otherwise it has nothing to get a time from.

edkraus, 2024-09-04:
Interesting read.  FWIW at another site (OHSU) that has a mix of QRT5, BB5, and SXT5 which I have set all to another local ntp server ... the SXT5s are behaving as we would expect and the QRT5 and BB5s are all off.

Ed

AI7NC, 2024-09-04:
Ok, I've moved 'update-clock' from /etc/cron.hourly to /etc

[image]

We are currently 26 min behind.

[image]

aanon4, 2024-09-04:
Could you add a `ps w` to check on ntpd? Thanks.

AI7NC, 2024-09-04:
[image]

AI7NC, 2024-09-04:
AI7BQ created a quick metric for us to track time loss vs our VM 'ha'. We'll post a follow up once we have more data but it's losing time pretty quickly.

[image]

AI7NC, 2024-09-04:
Seems we're losing about 22 seconds every 60 seconds real-time. 

[image]

AI7NC, 2024-09-05:
[image]

[image]

aanon4, 2024-09-05:
Probably time to run the ntpd command by hand (see earlier in this issue for specifics) and log the output of that to see what it's doing.

AI7NC, 2024-09-05:
[image]

After sync
[image]

Offset tracker indicating returning to offset of zero then continuing loss at ~22s every 60s [slope figure].
[image]

aanon4, 2024-09-05:
The ntp client will continue to run and print debugging, so the output over time would be useful. If you leave it running like this over night hopefully you can paste the entire log here tomorrow.

AI7NC, 2024-09-05:
Ok, I'll restart it and let it run. No promises as we're accessing this node through a backup link that has short drops every few hours so the SSH session may not be stable for 8 hours. If it breaks we'll retry it over a shorter period.

AI7NC, 2024-09-05:
I'll have it write to a log file so we have everything up until it disconnects if it does.

AI7NC, 2024-09-05:
Output so far. Looks like root distance/dispersion guarding may be the key to why it stops syncing after initial. 

root@W7PXL-LCCORE-EAST:/etc/init.d# /usr/sbin/ntpd -n -N -ddd -S /usr/sbin/ntpd-hotplug -p lccore-pbx.local.mesh
ntpd: 'lccore-pbx.local.mesh' is 10.104.67.130
ntpd: sending query to 10.104.67.130
ntpd: reply from 10.104.67.130: offset:+48.947045 delay:0.019047 status:0x24 strat:3 refid:0x68839baf rootdelay:0.027619 reach:0x01
ntpd: no valid datapoints, no peer selected
ntpd: poll:1s sockets:0 interval:1s
ntpd: sending query to 10.104.67.130
ntpd: reply from 10.104.67.130: offset:+49.557731 delay:0.020180 status:0x24 strat:3 refid:0x68839baf rootdelay:0.027619 reach:0x03
ntpd: current time is 2024-09-04 23:38:48.422171
ntpd: setting time to 2024-09-04 23:39:37.979902 (offset +49.557731s)
ntpd: executing '/usr/sbin/ntpd-hotplug step'
ntpd: update from:10.104.67.130 offset:+0.000000 delay:0.020180 jitter:0.002000 clock drift:+0.000ppm tc:5
ntpd: poll:32s sockets:0 interval:32s
ntpd: sending query to 10.104.67.130
ntpd: reply from 10.104.67.130: offset:+19.227141 delay:0.016823 status:0x24 strat:3 refid:0x68839baf rootdelay:0.027619 reach:0x07
ntpd: peer 10.104.67.130 unfit for selection: root distance 6.403275 too high, jitter:6.347738
ntpd: no valid datapoints, no peer selected
ntpd: poll:32s sockets:0 interval:32s
ntpd: sending query to 10.104.67.130
ntpd: reply from 10.104.67.130: offset:+38.457501 delay:0.020820 status:0x24 strat:3 refid:0x9bf8c41c rootdelay:0.055482 reach:0x0f
ntpd: peer 10.104.67.130 unfit for selection: root distance 13.422165 too high, jitter:13.362739
ntpd: no valid datapoints, no peer selected
ntpd: poll:32s sockets:0 interval:32s
ntpd: sending query to 10.104.67.130
ntpd: reply from 10.104.67.130: offset:+57.681998 delay:0.017391 status:0x24 strat:3 refid:0x9bf8c41c rootdelay:0.055482 reach:0x1f
ntpd: peer 10.104.67.130 unfit for selection: root distance 20.980473 too high, jitter:20.921862
ntpd: no valid datapoints, no peer selected
ntpd: poll:33s sockets:0 interval:32s
ntpd: sending query to 10.104.67.130
ntpd: reply from 10.104.67.130: offset:+77.519615 delay:0.023055 status:0x24 strat:3 refid:0x9bf8c41c rootdelay:0.055482 reach:0x3f
ntpd: peer 10.104.67.130 unfit for selection: root distance 28.634448 too high, jitter:28.572145
ntpd: no valid datapoints, no peer selected
ntpd: poll:33s sockets:0 interval:32s
ntpd: sending query to 10.104.67.130
ntpd: reply from 10.104.67.130: offset:+97.348398 delay:0.018875 status:0x24 strat:3 refid:0x9bf8c41c rootdelay:0.055482 reach:0x7f
ntpd: peer 10.104.67.130 unfit for selection: root distance 35.645511 too high, jitter:35.584472
ntpd: no valid datapoints, no peer selected
ntpd: poll:32s sockets:0 interval:32s
ntpd: sending query to 10.104.67.130
ntpd: reply from 10.104.67.130: offset:+116.577477 delay:0.021163 status:0x24 strat:3 refid:0x9bf8c41c rootdelay:0.055482 reach:0xff
ntpd: peer 10.104.67.130 unfit for selection: root distance 41.175844 too high, jitter:41.112891
ntpd: no valid datapoints, no peer selected
ntpd: poll:32s sockets:0 interval:32s
ntpd: sending query to 10.104.67.130
ntpd: reply from 10.104.67.130: offset:+135.808554 delay:0.022430 status:0x24 strat:3 refid:0x9bf8c41c rootdelay:0.055482 reach:0xff
ntpd: peer 10.104.67.130 unfit for selection: root distance 44.610819 too high, jitter:44.546470
ntpd: no valid datapoints, no peer selected
ntpd: poll:33s sockets:0 interval:32s
ntpd: sending query to 10.104.67.130
ntpd: reply from 10.104.67.130: offset:+155.635911 delay:0.016019 status:0x24 strat:3 refid:0x9bf8c41c rootdelay:0.055482 reach:0xff
ntpd: peer 10.104.67.130 unfit for selection: root distance 44.725198 too high, jitter:44.663239
ntpd: no valid datapoints, no peer selected
ntpd: poll:32s sockets:0 interval:32s
ntpd: sending query to 10.104.67.130
ntpd: reply from 10.104.67.130: offset:+174.866540 delay:0.025082 status:0x24 strat:3 refid:0x9bf8c41c rootdelay:0.055482 reach:0xff
ntpd: peer 10.104.67.130 unfit for selection: root distance 44.747016 too high, jitter:44.679768
ntpd: no valid datapoints, no peer selected
ntpd: poll:33s sockets:0 interval:32s
ntpd: sending query to 10.104.67.130
ntpd: reply from 10.104.67.130: offset:+194.691716 delay:0.020534 status:0x24 strat:3 refid:0x9bf8c41c rootdelay:0.055482 reach:0xff
ntpd: peer 10.104.67.130 unfit for selection: root distance 44.777976 too high, jitter:44.712203
ntpd: no valid datapoints, no peer selected
ntpd: poll:67s sockets:0 interval:64s
ntpd: sending query to 10.104.67.130
ntpd: reply from 10.104.67.130: offset:+234.943542 delay:0.021273 status:0x24 strat:3 refid:0x9bf8c41c rootdelay:0.055482 reach:0xff
ntpd: peer 10.104.67.130 unfit for selection: root distance 48.971495 too high, jitter:48.903495
ntpd: no valid datapoints, no peer selected
ntpd: poll:64s sockets:0 interval:64s

aanon4, 2024-09-05:
So back in October 2022, the OpenWRT folk applied a patch for the Basebox-5 to change the base clock frequency because, it was claimed, it was running to fast. I'm going to try reverting that patch and see what difference it makes (I have a basebox-5 handy to try this on). Back of the envelop math suggests the change in about what we're seeing here.

aanon4, 2024-09-05:
Doing a little more reading, and it looks like the change was for the Basebox-2 (the 2.4GHz version) so it's possible that the 5 GHz version we're looking at (and the QRT5 which is also based off the same low level configuration) still used the original frequency.

I do have an mANT 19s in service (921GS-5HPacD-19s) which is a newer AC device and seems to keep time just fine. Is the mANT 19s referenced in this bug a different version?

AI7NC, 2024-09-05:
We have an mANT 19s on the hospital, do you want to do the same drift test on it and see if the drift is the same or different?

AI7NC, 2024-09-05:
The specific basebox 5 we've been testing on is RB912UAG-gHPnD-OUT-US

AI7NC, 2024-09-05:
I'll have John set up the same metrics for the sector and a QRT so we can test if the drift rate is the same.

aanon4, 2024-09-05:
I ran my Basebox-5 overnight with the fix and the clock is still correct this morning. So that looks good.

What is the specific Mikrotik model number for the mANT 19s you 're using?

AI7NC, 2024-09-05:
Per John our sector is the MTAS-5G-15D120

aanon4, 2024-09-05:
That's an antenna without a radio. Is the radio also a basebox 5?

AI7NC, 2024-09-05:
You know it could be, let me ask John. Our original BOM indicated the 19s but I'm guessing they may have changed the design. The interface for the device indicates it's a basebox5.

aanon4, 2024-09-05:
https://github.com/aredn/aredn/pull/1462

edkraus, 2024-09-05:
Hi folks,
Is the clock fix also present in the 0905-b nightly I see for the QRT5?
Ed

AI7NC, 2024-09-05:
I don't believe so Ed, he likely did a local patch to test. [Here](https://github.com/aredn/aredn/pull/1462) is the pull request and it's not approved yet.

edkraus, 2024-09-05:
Thank you.  For tomorrow I'll attempt to put 3.24.6.0 back on the malfunctioning QRT5 and wait for more information.
Ed

aanon4, 2024-09-05:
The change will be in tomorrow build.

edkraus, 2024-09-05:
it will be labeled 0906 and released tonight, that's what you mean by "tomorrow"?

aanon4, 2024-09-05:
The build kicks off about 08:00 UTC Friday and will be 0906.

AI7NC, 2024-09-05:
Hi Tim,

Looks like we're set. We measured the same differential slope for all 3 devices, the both baseboxes and the QRT. So the offset adjustment should be the same. 

[image]

aanon4, 2024-09-05:
If you folk are able to test tomorrow's build on your devices to confirm, that would be fantastic.
And thank you for all the time you've put into diagnosing this problem; it's greatly appreciated.

AI7NC, 2024-09-06:
Sure thing, we'll update to nightly tomorrow and give it a shot.

AI7NC, 2024-09-06:
So far so good, we've loaded the nightly on 2 basebox 5s and we no longer see the deviation slope and they're holding the correct time. We're working to test it on a QRT or two and maybe wait 24 hours or so before releasing this issue to close.

[image]

edkraus, 2024-09-06:
I loaded the 0906 on the QRT5 that was giving us trouble and it's been working for the last several hours.  It has ntp pointed at a local server and the time it reports seems good.  I'll let you know if any new issues arise.
Ed

 
 

image.png (view on web)

AI7NC, 2024-09-06:
Thanks Ed, please give us an update late tonight or tomorrow morning if it
remains synced or has drifted.

73,

Nathan Campbell, AI7NC

AI7NC, 2024-09-07:
Moving to close this issue as I show it as resolved on both basebox5 and QRT5 hardware with the 9/6 nightly. Time appears solid for > 24 hours after the update via either internal or external NTP servers.

[image]

closed 2024-09-07 by AI7NC