aredn/aredn #1465: Max modulation setting in the UI

AI7NC, 2024-09-06:
**Is your feature request related to a problem? Please describe.**
We have some nodes with less than ideal connections. Usually pertaining to partial line of sight through foliage. These links while having decent SNR exhibit swings in packet delivery success and thus swing wildly through modulation rates over time leading to poor latency and jitter.

**Describe the solution you'd like**
I would like to have a UI setting that caps the maximum modulation for a device, thus locking us into a lower throughput yet higher probability of successful packet delivery. This would help us prioritize latency and jitter over maximum throughput for these types of links.

**Describe alternatives you've considered**
We are already at a bandwidth of 10 MHz, going lower has issues and not all devices on some of these sectors have line of sight issues.

**Additional context**

aanon4, 2024-09-06:
I'd very much like to do this, but when I've tinkered with the command options involved I've gotten very uneven feedback. Often it seems like it does nothing at all. Other times it seems like the described commands are wrong. I don't know if it's an unimplemented feature of the various different chipsets on radios, poor documentation, or me not understanding what's actually involved.

If you want to experiment then https://wireless.wiki.kernel.org/en/users/documentation/iw is a good starting point to the commands and options, but it's also a bit out of date.

aanon4, 2025-04-21:
Closing - no feedback received.