aredn/aredn #1467: unable to open DHCP settings when DHCP is disabled

ko0ooo, 2024-09-06:
### Describe the bug
One of my nodes is configured as an access point for my cell phone.   The phone gets its IP from my primary node, so the DHCP on the AP is disabled.

There may come a time when I'll need to bypass my network and talk directly to that node.  Having it issue an IP would help.

After logging in to the AP using the new UI, I'm unable to access the LAN settings to activate the DHCP.  I can when reverting to the old UI.   If the AP is configured as DHCP, no problems accessing the LAN settings.

thanks...

Richard     ko0ooo

### Expected behavior
Be able to configure/modify the LAN settings when DHCP is disabled.

### Screenshots
[image]

### Additional context
Add any other context about the problem here.

aanon4, 2024-09-06:
You need to enable the LAN which is in the Network panel.

aanon4, 2024-09-06:
http://docs.arednmesh.org/en/latest/arednGettingStarted/node_admin.html#network-settings

ko0ooo, 2024-09-07:
that was it..    thanks...

Richard    ko0ooo

closed 2024-09-07 by ko0ooo