aredn/aredn #1473: Port fowarding in new UI does not work

kodiakastu, 2024-09-07:
### Describe the bug
Hi. I tried updating to latest nightly (20240907) and I'm running a VM. When I add a port forward in new ui, it does not show in new ui again, even after rebooting. It does show in old ui, but it's not working. After deleting the entry and redoing it in old ui, it works as expected and shows in new ui.

### Expected behavior
Navigate to wanip:exposedport and service should work.

### Additional context
Here is support data when it's in between states.
[supportdata-VA2LRM-LPLN-Q01-VM00-2024-9-7-19-12.tar.gz](https://github.com/user-attachments/files/16920122/supportdata-VA2LRM-LPLN-Q01-VM00-2024-9-7-19-12.tar.gz)
m here.

aanon4, 2024-09-07:
https://github.com/aredn/aredn/pull/1474

aanon4, 2024-09-07:
Actually, https://github.com/aredn/aredn/pull/1476 is a better way to do this as it keeps the WAN visual but provide the wan value.

kodiakastu, 2024-09-07:
Will let you know tomorrow. Thanks for the swift answer!

kodiakastu, 2024-09-08:
Fix seems good to me! Thanks again.

closed 2024-09-08 by kodiakastu