aredn/aredn #1480: mac-address texy box to short

KQ6EO, 2024-09-10:
### Describe the bug
[KQ6EO-HAP-TSTC admin.pdf](https://github.com/user-attachments/files/16940153/KQ6EO-HAP-TSTC.admin.pdf)

Sorry I accidently closed the original bug report

The new GUI's mac-address text box on the "LAN DHCP" page is too short and cuts off the last character or two of the mac-address.
I am using the Firefox browser.

aanon4, 2024-09-10:
https://github.com/aredn/aredn/issues/1479
You could have just reopened it - but not to worry.

aanon4, 2024-09-10:
Anyway, I'm not sure what the PDF image is of, but it's like you have no CSS loaded for some reason.

aanon4, 2024-09-10:
[image]
This is what mine looks like.

aanon4, 2024-09-10:
Maybe you used the print feature to get that "screenshot"?  I'm going to need a real screenshot.

K5RA, 2024-09-10:
You might want to review closed issues #1371, #1376, and #1389 which address this.

KQ6EO, 2024-09-10:
[image]

The problem is only happening on the "High-Contrast" theme.  As K5RA noted you can click in the text box of the "Address Reservations" section and move right to see the rest of the mac address.  This does not work in the "Active Leases" section.

K5RA, 2024-09-10:
Double clicking on either IP adr or mac adr in Active Leases allows to you see either end of the value by using the right and left direction arrow keys.  Easiest way is to use Light Theme if you are going to be in the window a while.

aanon4, 2024-09-10:
> The problem is only happening on the "High-Contrast" theme.
Ah - you should have said at the beginning. Yes, the High Contrast theme makes text bigger and in some cases it wont fit as well. It's a trade off as the dialog sizes are fixed. I will continue to tweak things a little to try to improve these compromises.

closed 2024-09-10 by aanon4