aredn/aredn #1482: Feature Request; Metric units

Linus-007, 2024-09-10:
Is it possible to write in user-selectable settings for units of measure? International adoption might be easier. I sure wish I could see the distances in km.

kodiakastu, 2024-09-13:
I don't wanna beat a dead horse, but if in your browser response to `request.getLocale` "en-us" is in the list, where ever it is, the Ui will show imperial units. I had to find in FF the list of preferred languages to get rid of en-us.

aanon4, 2024-09-13:
Oh I'm all for beating dead horses.
Is there a better way for me to detect the actual locale of the browser if "en-US" is stuck places is really shouldnt be?

kodiakastu, 2024-09-13:
I had a long list of preferred languages and en-us was one of them. The browser is supposed to feed them in the preferred order. I'm not that good in programming; maybe just checking the index 0 in the object ? 

Here's a screen shot from what I had from https://www.localeplanet.com/support/browser.html:
[image]

I think en-us is there by default on FF.

aanon4, 2024-09-13:
I'd be interested in the server side info (this is client side). Send over the output from the following:

```http://localnode.local.mesh/a/info```

kodiakastu, 2024-09-13:
That's my FF (he's behaving correctly):
```
request.env.PATH_INFO: /info
request.env.HTTP_ACCEPT: text/html,application/xhtml+xml,application/xml;q=0.9,image/avif,image/webp,image/png,image/svg+xml,*/*;q=0.8
request.env.HTTP_ACCEPT_ENCODING: gzip, deflate
request.env.HTTP_ACCEPT_LANGUAGE: en-CA,fr-CA;q=0.7,en;q=0.3
request.env.HTTP_CONNECTION: keep-alive
request.env.HTTP_COOKIE: authV1=cm9vdDokMSRvYnMvdU9DdSRaMzJBTFRaYUpvdkJTcFprRU9LQjMvOjE5OTY2OjA6OTk5OTk6Nzo6Og==; meshchat_id=2gTLu; meshchat_call_sign=VA2LRM
request.env.HTTP_HOST: 172.19.14.33:8080
request.env.HTTP_USER_AGENT: Mozilla/5.0 (Windows NT 10.0; Win64; x64; rv:129.0) Gecko/20100101 Firefox/129.0
request.env.GATEWAY_INTERFACE: CGI/1.1
request.env.SERVER_SOFTWARE: uhttpd
request.env.SCRIPT_NAME: /a
request.env.SCRIPT_FILENAME: /app/root.ut
request.env.QUERY_STRING: 
request.env.REQUEST_URI: /a/info
request.env.SERVER_PROTOCOL: HTTP/1.1
request.env.REQUEST_METHOD: GET
request.env.REDIRECT_STATUS: 200
request.env.SERVER_NAME: 172.19.14.33
request.env.SERVER_ADDR: 172.19.14.33
request.env.SERVER_PORT: 8080
request.env.REMOTE_HOST: 172.19.13.5
request.env.REMOTE_ADDR: 172.19.13.5
request.env.REMOTE_PORT: 61315
request.env.HTTP_VERSION: 1.1
request.headers.URL: /a/info
request.headers.host: 172.19.14.33:8080
request.headers.user-agent: Mozilla/5.0 (Windows NT 10.0; Win64; x64; rv:129.0) Gecko/20100101 Firefox/129.0
request.headers.accept: text/html,application/xhtml+xml,application/xml;q=0.9,image/avif,image/webp,image/png,image/svg+xml,*/*;q=0.8
request.headers.accept-language: en-CA,fr-CA;q=0.7,en;q=0.3
request.headers.accept-encoding: gzip, deflate
request.headers.connection: keep-alive
request.headers.cookie: authV1=cm9vdDokMSRvYnMvdU9DdSRaMzJBTFRaYUpvdkJTcFprRU9LQjMvOjE5OTY2OjA6OTk5OTk6Nzo6Og==; meshchat_id=2gTLu; meshchat_call_sign=VA2LRM
request.headers.upgrade-insecure-requests: 1
request.headers.priority: u=0, i
uhttpd.send: function send(...) { [native code] }
uhttpd.sendc: function send(...) { [native code] }
uhttpd.recv: function recv(...) { [native code] }
uhttpd.urldecode: function urldecode(...) { [native code] }
uhttpd.urlencode: function urlencode(...) { [native code] }
uhttpd.docroot: /www
```
and here's my edge (who doesnt behave):
```
request.env.PATH_INFO: /info
request.env.HTTP_ACCEPT: text/html,application/xhtml+xml,application/xml;q=0.9,image/avif,image/webp,image/apng,*/*;q=0.8,application/signed-exchange;v=b3;q=0.7
request.env.HTTP_ACCEPT_ENCODING: gzip, deflate
request.env.HTTP_ACCEPT_LANGUAGE: fr-CA,fr;q=0.9,fr-FR;q=0.8,en-CA;q=0.7,en;q=0.6,en-US;q=0.5,en-GB;q=0.4,pt;q=0.3
request.env.HTTP_CONNECTION: keep-alive
request.env.HTTP_HOST: 172.19.14.33
request.env.HTTP_USER_AGENT: Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/128.0.0.0 Safari/537.36 Edg/128.0.0.0
request.env.GATEWAY_INTERFACE: CGI/1.1
request.env.SERVER_SOFTWARE: uhttpd
request.env.SCRIPT_NAME: /a
request.env.SCRIPT_FILENAME: /app/root.ut
request.env.QUERY_STRING: 
request.env.REQUEST_URI: /a/info
request.env.SERVER_PROTOCOL: HTTP/1.1
request.env.REQUEST_METHOD: GET
request.env.REDIRECT_STATUS: 200
request.env.SERVER_NAME: 172.19.14.33
request.env.SERVER_ADDR: 172.19.14.33
request.env.SERVER_PORT: 80
request.env.REMOTE_HOST: 172.19.13.5
request.env.REMOTE_ADDR: 172.19.13.5
request.env.REMOTE_PORT: 61246
request.env.HTTP_VERSION: 1.1
request.headers.URL: /a/info
request.headers.host: 172.19.14.33
request.headers.connection: keep-alive
request.headers.upgrade-insecure-requests: 1
request.headers.user-agent: Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/128.0.0.0 Safari/537.36 Edg/128.0.0.0
request.headers.accept: text/html,application/xhtml+xml,application/xml;q=0.9,image/avif,image/webp,image/apng,*/*;q=0.8,application/signed-exchange;v=b3;q=0.7
request.headers.accept-encoding: gzip, deflate
request.headers.accept-language: fr-CA,fr;q=0.9,fr-FR;q=0.8,en-CA;q=0.7,en;q=0.6,en-US;q=0.5,en-GB;q=0.4,pt;q=0.3
uhttpd.send: function send(...) { [native code] }
uhttpd.sendc: function send(...) { [native code] }
uhttpd.recv: function recv(...) { [native code] }
uhttpd.urldecode: function urldecode(...) { [native code] }
uhttpd.urlencode: function urlencode(...) { [native code] }
uhttpd.docroot: /www
```

aanon4, 2024-09-13:
Can you see `http://localnode.local.mesh/a/status` and if so, what distance unit are the neighbor nodes in now?

kodiakastu, 2024-09-13:
Unfortunately, I do not have acces to a mesh hooked to one of the super nodes. I checked with my neighbour (wg tun) and it is consistant to what i have localy.

aanon4, 2024-09-13:
Okay - try this IP address instead (I'm opening the WAN port on the local supernode for a moment to test this)
206.197.44.10

kodiakastu, 2024-09-13:
With yours, both are showing in metric

aanon4, 2024-09-13:
https://github.com/aredn/aredn/pull/1496

aanon4, 2024-09-13:
Reopending as it was broken. If you can verify this tomorrow that'd be lovely.

kodiakastu, 2024-09-13:
as usual. Thanks!

kodiakastu, 2024-09-13:
Works as advertised! Defaults to metric unless a US language is present!

closed 2024-09-13 by aanon4

Linus-007, 2024-09-24:
This is unclear from the documentation and a very odd way to set the metric/imperial preferences. Some of us in the USA prefer the metric system, and Some in metric countries prefer the imperial measurement.