aredn/aredn #1484: Icon misalignment

Orv, 2024-09-10:
### Describe the bug
A clear and concise description of what the bug is.
Action icon misaligned with its box (see screenshot)
Note - only happens with FireFox 130.0 on Kubuntu 24.04.1.  Does not happen with Chrome 126.0.6478.182 on same system
- [ ] Does this issue still occur on the latest version? (**Required**)
Yes
### Expected behavior
A clear and concise description of what you expected to happen.
Same behavior as viewing on other browsers - icons properly aligned.

[image]

aanon4, 2024-09-10:
One if my favorite games .. guess the subtle css difference between implementations of the same browser but on different platforms :-)

Orv, 2024-09-10:
Steve couldn't reproduce it on Firefox 130.0 on Debian.  You can close this as WontFix if you want.

aanon4, 2024-09-10:
Naw .. I fixed it with a magic "no idea why this works really" css change.
Rolled into this one - https://github.com/aredn/aredn/pull/1481