aredn/aredn #1485: Firmware Update to Mikrotik hAP Lite reports "Error: Unknown Error"

km6aow, 2024-09-10:
### Describe the bug
When updating firmware, an unknown error is flagged, but the update proceeds.

- [ ] Does this issue still occur on the latest version? (**Required**) 
- [ ] I'm on the current nightly build.

### Expected behavior
We shouldn't see a red error banner for a successful update.

### Screenshots
[image]

### Additional context
This hAP is downloading firmware from a mirror if that makes any difference.

aanon4, 2024-09-10:
Please attached support data - thanks.

aanon4, 2024-09-10:
I stuck your local repo into as the download source for a node here, and it worked just fine. So I'm going go assume something network related is going on.

km6aow, 2024-09-10:
No worries. If it works elsewhere, I can live with the "warning" as the update does in fact succeed. Thanks!

closed 2024-09-10 by km6aow

km6aow, 2024-09-10:
The computer from which I did that was WiFi connected and has multiple APs it knows about here. As soon as the hAP drops its WiFi signal, that machine automatically rejoins another AP in the house. It never does get to the "updating" screen.