aredn/aredn #1489: Error Popup viewing Tunnels on HAP AC3 in new UI

aa3jc, 2024-09-11:
### Describe the bug
A clear and concise description of what the bug is.
On the last 4 Nightly builds in the new UI, when logged in as Admin , if I click on "Tunnels" I get a Error Popup window over the node window.  The only way to clear is to refresh the browser, but cannot view, change, or add tunnels.  This happens on the last 4 Nightly builds.

- [ ] Does this issue still occur on the latest version? (**Required**)  Yes and the previous 4 builds.

### Expected behavior
A clear and concise description of what you expected to happen.
Would like to view, modify, add, or delete tunnels.

### Screenshots
Please include a screenshot of your system information (if possible) if the specific system environment is relevant to the bug.
See the attached two screenshots.
[Screenshot Error Tunnels.pdf](https://github.com/user-attachments/files/16971184/Screenshot.Error.Tunnels.pdf)
[Screenshot of Node Info.pdf](https://github.com/user-attachments/files/16971186/Screenshot.of.Node.Info.pdf)

If applicable, add screenshots to help explain your problem.

### Additional context
Add any other context about the problem here.

aanon4, 2024-09-12:
Please attach the support data - thanks

aanon4, 2024-09-12:
The error looks like it would be generated by an invalid tunnel start ip address. The support data will confirm this.

aa3jc, 2024-09-12:
[supportdata-AA3JC-AC3-HATFLD-1-2024-9-12-5-21.tar.gz](https://github.com/user-attachments/files/16976705/supportdata-AA3JC-AC3-HATFLD-1-2024-9-12-5-21.tar.gz)

aanon4, 2024-09-12:
In the file /etc/config.mesh/vtun I see the following lines:
```
config network
        option start '172.31.80.00'
        option dns 'www.aa3jc.com'
```
Note the 00 at the end of the 'start' value. This is invalid and should be a single 0. If you change this to  a single 0 all should be well again. I'm not sure how it got like this, but possibly the old UI allowed it (it should not have). On the plus side, the old UI will let you fix it.

aa3jc, 2024-09-12:
That fixed it! Didnt even have to reboot!!

Thanks for the quick fix!
Joe-AA3JC

closed 2024-09-12 by aa3jc