aredn/aredn #1491: Installed Packages Count Incorrect after Nightly Download

K5RA, 2024-09-12:
Downloaded N 20240912 this morning.  About an hour later (node idle), checked Status and saw that Installed Packages count was zero despite Meshchat being installed and running.  Opened then closed Firmware area.  Installed Packages count now one.  Is this expected?  

I would prefer Installed Packages count to be correct after node recovers from reboot.  

System: hAP ac lite, latest nightly.  RP5B SBC with updated Bookworm and Chromium.

aanon4, 2024-09-12:
Please attach the support data - thanks.

K5RA, 2024-09-12:
[supportdata-K5RA-HAP-FB-2024-9-12-10-45.tar.gz](https://github.com/user-attachments/files/16982252/supportdata-K5RA-HAP-FB-2024-9-12-10-45.tar.gz)

aanon4, 2024-09-12:
https://github.com/aredn/aredn/pull/1493

k1ky, 2024-09-13:
I have experienced this as well on many occasions.  It corrects itself after time or revisiting installed packages screen.

aanon4, 2024-09-13:
Perhaps I'm overthinking this - is the issue that the package count is 0 shortly after update before the browser is refreshed; but is then correct after a page refresh?

K5RA, 2024-09-13:
Yes.  The package count was zero when node rebooted with new nightly this morning.  When I open/closed the Packages window, package count went to 1.

Out of curiosity, I downloaded N 20240913 on top of N 20240913 just now.  Installed packages count came up zero on reboot.  Refreshing the screen by clicking AREDN logo changed it to one.  Easy workaround.

closed 2024-09-13 by aanon4