aredn/aredn #1495: Special Characters in description field sometimes not interpreting properly

k1ky, 2024-09-13:
### Describe the bug
A clear and concise description of what the bug is.
Special Character "'" for feet in description isn't always interpreted properly in new UI. N-2024-0912 and maybe a few nightlies before this.

- [ ] Does this issue still occur on the latest version? (**Required**)
- [ ] Yes - N-20240912

### Expected behavior
A clear and concise description of what you expected to happen.
Expect special characters to display properly.

### Screenshots
Please include a screenshot of your system information (if possible) if the specific system environment is relevant to the bug.
[image]
[supportdata-KL7AA-SUMMIT-MANT19s-25MST310-2024-9-12-17-46 (1).tar.gz](https://github.com/user-attachments/files/16987453/supportdata-KL7AA-SUMMIT-MANT19s-25MST310-2024-9-12-17-46.1.tar.gz)

If applicable, add screenshots to help explain your problem.

### Additional context
Add any other context about the problem here.

k1ky, 2024-09-13:
I "believe" that I added this description information from the new UI GUI side after initial firmware install.  It shows the same graphical character on the Old UI side as well.

aanon4, 2024-09-13:
Yeah. I dont escape much in the description text. Just enough so it doesnt break the config file. I need to look for a definitive list of characters I need to care about.

k1ky, 2024-09-13:
I use ', (, ) and | mostly - nothing too fancy

aanon4, 2024-09-13:
But < and > dont work either - so I need a proper list. I just need to look for it.

aanon4, 2024-09-13:
> I use ', (, ), @, - and | mostly - nothing too fancy

Which of these doesnt work?

k1ky, 2024-09-13:
So far only "'" foot mark is all that I have seen so far.

k1ky, 2024-09-13:
Interesting that it shows okay in the description entry box but not on the display.
Entry:
[image]
Display:
[image]

k1ky, 2024-09-13:
Seems to be some issue during firmware upgrade.  I am able to edit it and displays fine.

k1ky, 2024-09-13:
Confirmed that translation is okay when going from Nightly 08 to 12, will investigate further if needed.

aanon4, 2024-09-13:
I can do better though
https://github.com/aredn/aredn/pull/1497

kodiakastu, 2024-09-14:
It's not just about some special caracters. This doesn't fix french diacritics and leave other things broken. 

If I save thru old UI I can get away with a lot of weird things, even emojis. As soon I change something in new UI it breaks everything. When it breaks, a lot of Â appears. 

I tried the opposite way: I save something in new ui, is fine, but when I reload the page it's broken again, but not in the dialog box and is broken in old ui. 

What changed between old and new UI that requires to change the way we store the node's description?

I used ``- ⚠➡🏠😊😂🤣 - !"/$?&*()_+#|\<>{};:`,.'^ - ±@£¢¤¬¦²³¼½¾¸¨ - à é ê è`` as a test string. Let me know if it's too much ;)

aanon4, 2024-09-14:
So this is a different bug. The fixes before were to escape character which are problematic for HTML to show because they means something special (< and > for example). This is just a utf8 encoding issue. Supposedly HTML5 is utf8 by default so I shouldnt need to specify it .. but that seems not to be the case. So ...
https://github.com/aredn/aredn/pull/1504

kodiakastu, 2024-09-14:
Was this early enough to be included in today's nightly?

aanon4, 2024-09-14:
Looks like the automated build failed last night so running again this morning. Will be available in a couple of hours if you check by hand.

kodiakastu, 2024-09-14:
#1504 Fixes new ui but breaks old ui

aanon4, 2024-09-14:
No plans to fix the old UI.

kodiakastu, 2024-09-14:
I also checked sysinfo.json: it displays correctly when I save in new UI.

I don't know if the description is used somewhere else.

closed 2024-09-14 by aanon4