aredn/aredn #1500: Clicking 'whole AREDN mesh' button isnt selecting closest or quickest supernode

KD4WLE, 2024-09-13:
### Describe the bug
Clicking 'whole AREDN mesh' button isnt selecting closest or quickest supernode. Both the local node and the super node are on the same VM and show as 'Local Nodes' , however the localnode picks a Supernode that shows as 'Fair'(2.2), bypassing both my own supernode and another tagged as 'Good'(1.2)

For good mesure, I have also check this from a node outside of the VM with the same unexpected result.

- [x] Does this issue still occur on the latest version? (**Required**)
- [ Yes, Both Supernode and node are on latest nightly] 

### Expected behavior
Should select the closest and/or fastest node

### Screenshots
Please include a screenshot of your system information (if possible) if the specific system environment is relevant to the bug.

[image]

### Additional context
Add any other context about the problem here.

aanon4, 2024-09-13:
Please attach the support data for the node.

aanon4, 2024-09-13:
And you cant attach it by email.

KD4WLE, 2024-09-13:
Well then.. 

[supportdata-KD4WLE-Titusville-One-2024-9-13-20-49.tar.gz](https://github.com/user-attachments/files/16999190/supportdata-KD4WLE-Titusville-One-2024-9-13-20-49.tar.gz)

aanon4, 2024-09-13:
You probably need to give me telnet access I'm afraid - nothing obvious looks wrong (clearly it is though).

KD4WLE, 2024-09-13:
Enabled.

aanon4, 2024-09-13:
You also need to send me the password (email address is in my github profile)

aanon4, 2024-09-13:
Thanks
https://github.com/aredn/aredn/pull/1503

KD4WLE, 2024-09-14:
Seems to be resolved. Thank you!

closed 2024-09-14 by KD4WLE