aredn/aredn #1514: New UI corrupts page layout

m0xzl, 2024-09-16:
### Describe the bug
Dashboard left open in browser for several hours.
The layout of the dashboard sections is corrupted over time with elements posted into different areas of the page.
A page reload resolves the situation.

### Expected behavior
The UI dashboard should periodically refresh and be presented with the same layout every time.

### Screenshots
[image]

### Additional context
Firmware version = 20240915-84824e8. The problem was also observed on a node running an earlier firmware version (nightly build - several days ago)
This screenshot context: Windows 10 / Edge
Previously observed context: Raspberry Pi (Bookworm) / Chrome

aanon4, 2024-09-16:
Interesting. So the way this panel works is that pieces of it are updated in the background so long as the window is the active tab in the browser. But this looks like rather than fetching just be piece of information it needs, it's pulling the entire page. As I dont have an obviously mechanism for this, I wondered if you could provide a way to replicate this in much more detail than you have? I booted up an AR750, put the current nightly on it, and have given it an hour but nothing much has happened so far. Is there anything in particular I need to do? Thanks.

aanon4, 2024-09-16:
https://github.com/aredn/aredn/pull/1515

This is my best guess at what's going on; but it would be good to have a way to verify this fixes the issue.

m0xzl, 2024-09-16:
Hi Tim

 

I have only observed this twice. Once on an AR150, and the one today on an AR750.

I have quite a few nodes and have been following the New UI nightly builds on 6 of them. All have behaved consistently.

 

I usually leave a browser open on the node that runs my tunnel server and connects to the internet. When I glanced at it this afternoon I saw the image in the screenshot.

 

The previous time I saw this presentation was on an AR150 which was connected to the mesh but with nothing connected to it. Again I had been monitoring the FW upgrades and left a browser open on the nodes dashboard.

 

Both times it has happened when I wasn’t looking, but it was when the page had been left open in a browser for a long time.

 

Sorry, I can’t offer anything more definitive than that. I hope it doesn’t take too much effort to track it down.

 

Cheers…

aanon4, 2024-09-20:
Does this still happen?

m0xzl, 2024-09-20:
I haven't seen it again. OK to close this. I will report again if I do see it.

closed 2024-09-20 by aanon4