aredn/aredn #1517: New Wireguard tunnels duplicating ports

k6ccc, 2024-09-16:
### Describe the bug
My K6CCC-hAPac3-at-Home node has had five existing WireGuard tunnels as the server end essentially ever since WG tunnels became available.  They use ports 5525 - 5529 as expected.  Yesterday that node was upgraded to Nightly build 20240915, and last night, I added two additional WireGuard tunnels to that node using the new UI.  The two WG tunnels duplicated ports 5528 and 5529.  I was able to test one of the new tunnels and it was unable to connect.

[supportdata-K6CCC-hAPac3-at-Home-2024-9-16-13-47.tar.gz](https://github.com/user-attachments/files/17019800/supportdata-K6CCC-hAPac3-at-Home-2024-9-16-13-47.tar.gz)

- [ ] Does this issue still occur on the latest version? (**Required**)

Today, I updated this node to nightly 20230916, and observed the same condition.  Next, I deleted the two tunnels, Committed, and rebuilt them and Committed.
Again, ports 5528 and 5529 were duplicated.

[image]

[supportdata-K6CCC-hAPac3-at-Home-2024-9-16-14-0.tar.gz](https://github.com/user-attachments/files/17019832/supportdata-K6CCC-hAPac3-at-Home-2024-9-16-14-0.tar.gz)

### Expected behavior
Not duplicate port numbers.

### Screenshots
Included screen capture of part of the Tunnel listing showing the duplicates along with one of the existing (KI6GRC-HAP-1 is one of the existing)

### Additional context

There are two support file included.  2024-9-16-13-47 is before making additional changes today.  2024-9-16-14-0 is after updating firmware and deleting and rebuilding the two tunnels.

aanon4, 2024-09-16:
Ports and IP addresses are assigned together, so this is happening because you have these two IP addresses 172.31.232.4 and 172.31.231.4 when it should not be possible for both of them to exist in the same tunnel configuration.
I see the tunnel start is set to 172.31.230.248 which means we should only ever see 172.31.230.X and 172.31.231.X addresses. So how come there are 172.31.232.X addresses? Are these the new ones (it's not clear from your report which are the old ones and which are the new ones)? Or did the start address get changed at some point?

aanon4, 2024-09-16:
I can at least stop the value being changed once tunnels exist.
https://github.com/aredn/aredn/pull/1518

k6ccc, 2024-09-16:
I agree that the IP addresses also did not make total sense either, but it was the duplicated ports that caught my attention.  In the screen capture, the two bottom ones shown - K6CCC-AR750-Portable and K6CCC-HAP-Portable and the duplicated ports.  I have never manually over ridden the auto-assigned IP addresses for anything in any node.  So this is what the node assigned.
Correction on what you typed, the first WG tunnel is 172.31.231.248, and continuing with .250 and .252.  I would have expected the next original WG tunnel to be 172.31.232.2, however AJ6GZ and KI6GRC went back to 172.31.231.2 and .4.  However those have worked since they were created shortly after WG tunnels were introduced.  That may have not been the correct IP addresses, but that is what the node built.  That may have been an old bug - I don't know.  However when I created K6CCC-HAP-Portable and K6CCC-AR750-Portable last night, it appears that it may have correctly continued the IP assignment order starting with 172.31.232.2 and .4.  Not that it should make any difference, but both K6CCC-AR750-Portable and K6CCC-HAP-Portable previously had legacy tunnels that I had never gotten around to converting to WG.  Last night, I deleted the old legacy tunnels and built them as new WG tunnels.

aanon4, 2024-09-16:
The start address is set to 172.31.230.248 which means legacy tunnels start at 172.31.230.248 and wireguard tunnels start at 172.31.231.248

Sorry - I was wrong; the 172.31.231.X and 172.31.232.X addresses can co-existing - got confused about how we wrap this stuff.

And I think this is the crux of the problem as you note above. How I wrap the allocation of IP:PORTS has changed as the UI has been debugged. Before I was wrapping it one way which was incompatible with the old UI, so it was updated to how it is now. Unfortunately you allocated two tunnels - 172.31.231.2 and .4 - while it was broken. The answer is to remove these two and reallocate.

*Hopefully* I have it right now. That'd certainly be nice.

k6ccc, 2024-09-16:
I was wondering if removing and rebuilding the two old (but working - wrong) was the solution.  I will do that and report.

k6ccc, 2024-09-17:
Deleted and rebuilt the two wrong tunnels and rebuilt them.  Looks like that is resolved.

closed 2024-09-17 by aanon4