aredn/aredn #1520: FTP server on a NODE.

mathisono, 2024-09-17:
I would like to utilize the USB port on hapAC, hapAC2, hapAC3. Im requesting a feature to detect a usb storage device, select it to be automatic formatted, mounted, and permissions  set for the attached USB storage device. to be The provide file system is then usable for a tiny ftp server running on the node or meshchat. 

-The mounted drive's file system would be available for the ftp server, and or mesh chat.
-internal FTP service. 
-A simple UI or Config file to define users for the ftp server. Preference is a config file.
-Read write permission for the file system for: FTP server and Meshchat.
-file system management should still be done thru the CLI! 
-The usb drive adoption should be unattended.

aanon4, 2024-09-17:
This sounds like an external package, though probably with dependencies we can build as part of the standard packages.

aanon4, 2024-10-04:
Recommend this is implemented as an external package.

closed 2024-10-04 by aanon4