aredn/aredn #1521: Unknown error encountered with firmware upgrade

m0xzl, 2024-09-17:
### Describe the bug
During a firmware update initiated from the New UI the modal stalls and becomes unresponsive.
* I use the Download Firmware option.
* The available firmware list loads OK.
* I selected the firmware version that I wanted. This appears to work OK.
* The "Fetch and Update" and "Done" buttons are not enabled.
* After a short delay the error message "ERROR: Unknown error" appears at the top.

### Expected behaviour
The "Fetch and Update" and "Done" buttons should be enabled and clicking on the "Fetch and Update" button should start the download process.

### Screenshots
[image]

### Additional context
* I have encountered this problem quite a few times. I believe that it has happened on several different nodes but I didn't note the node types. This time it is an AR150. Previous occurrences were most likely on AR150 and AR300 nodes.
* Resetting from this condition usually requires closing the browser tab and re-opening.
* I don't believe it is happening to the same node repeatedly. Usually it works as expected.
* I have pretty much been following the nightly updates since the New UI was released so I have done 'lots' of updates. Most work as expected. I have encountered this problem at least 6 times but didn't think of it as a bug until now.

aanon4, 2024-09-17:
A few questions:

> I selected the firmware version that I wanted. This appears to work OK.
> The "Fetch and Update" and "Done" buttons are not enabled.

So you select the firmware (I assume nightly) from the list offered, but after you do that the "Fetch and Update" and "Done" buttons do not get enabled?

> After a short delay the error message "ERROR: Unknown error" appears at the top.

So this bit suggests there were enabled, that you pressed "Fetch and Update" which then disabled it and the "Done" button, and - given the image included - the done button was re-enabled once the Error message was displayed.  Is this was is happening? (the Fetch.. button will not be re-enabled until you reselect something new to download).

Assuming the above, then the process is working as it's intended. Why the Error is a different question. I've had at least one report that when this happens the node is upgrading in the background anyway. Is this what happens here or something else?

m0xzl, 2024-09-17:
* Yes latest nightly selected in this case.
* On reflection, I am often doing the process on autopilot and it is possible that I have clicked on "Fetch and Update" before paying proper attention to the state.
* I was alerted to the problem by the fact that the progress bar remained empty.
* I appreciate that processes sometimes fail for a variety of reasons, and that's OK. The unknown error is more puzzling and I am not sure if I should just try again, or there is something more complicated going on.
* If this happens again I will attempt to take better notes.

aanon4, 2024-09-17:
During an update, the node maintains its connection to the browser to update it to the progress, the the browser can update the loading bar. I think the unknown error happens when this connection terminates unexpectedly.  If you do have it happen again, it would be useful to get the support data immediately after, in case something useful has been logged.

m0xzl, 2024-09-17:
Will do.

aanon4, 2024-09-17:
Hopefully this will improve the error reporting on various incomplete download errors
https://github.com/aredn/aredn/pull/1522

aanon4, 2024-10-04:
No feedback in 2 weeks.

closed 2024-10-04 by aanon4