aredn/aredn #1524: New : UI

mathisono, 2024-09-17:
When login into the new UI, with the wrong password, the UI docent prompt the user of a failed login attempt.  

With no distinct change to the UI after login, its not apparent that you've login or failed.

aanon4, 2024-09-17:
The visual changes when you login are described in the documentation here - http://docs.arednmesh.org/en/latest/arednGettingStarted/node_admin.html
This is in addition to the secondary visual changes where more information is show (notable Internal Services).

As for explicitly flagging the failed login - I'll convert this to a feature request for that.

mathisono, 2024-09-17:
feature or bug... Its comes back to the design aesthetic. Black and gray dont offer an indication of a "state". The UI needs hypertext coloration, that consistent across styles.  I suggest Purple and Green. why? Purple is unused in the current UI, it an accessible color too. Green because is already associated with "Good" , "up to date" , "active" because of this it can too be associated with "login successful". Purple can be missed, and is distinct from other colors.  Even github uses a purples shade for "Something isn't working".

https://www.nceas.ucsb.edu/sites/default/files/2022-06/Colorblind%20Safe%20Color%20Schemes.pdf

closed 2024-09-19 by aanon4