aredn/aredn #1526: I2C bus doesn't work in GL-AR750 (cross-post from OpenWRT)

bwarden, 2024-09-17:
### Describe the bug

The I2C bus on the GL-AR750 fails to scan properly, falsely reporting a device on every possible address, regardless of whether any I2C devices are connected to the bus.

Patch is available upstream via OpenWRT.

- [x] Does this issue still occur on the latest version? (**Required**)

### Expected behavior
Using i2cdetect, I expect to see the addresses of only detected devices.

### Screenshots
Booting the GL-AR750, I can see that the gpio-based I2C controller is set up with pins 16 and 17:
```
[    0.147948] gpio-17 (sda): enforced open drain please flag it properly in DT/ACPI DSDT/board file
[    0.157395] gpio-16 (scl): enforced open drain please flag it properly in DT/ACPI DSDT/board file
[    0.166869] i2c-gpio i2c: using lines 17 (SDA) and 16 (SCL)
[    1.644658] i2c_dev: i2c /dev entries driver
```
Attempting to scan the bus, with nothing connected, shows unexpected results matching all possible devices:
```
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:                         08 09 0a 0b 0c 0d 0e 0f
10: 10 11 12 13 14 15 16 17 18 19 1a 1b 1c 1d 1e 1f
20: 20 21 22 23 24 25 26 27 28 29 2a 2b 2c 2d 2e 2f
30: 30 31 32 33 34 35 36 37 38 39 3a 3b 3c 3d 3e 3f
40: 40 41 42 43 44 45 46 47 48 49 4a 4b 4c 4d 4e 4f
50: 50 51 52 53 54 55 56 57 58 59 5a 5b 5c 5d 5e 5f
60: 60 61 62 63 64 65 66 67 68 69 6a 6b 6c 6d 6e 6f
70: 70 71 72 73 74 75 76 77
```
The underlying issue is with the device tree configuration:
https://github.com/openwrt/openwrt/blob/2bec6f48e6f8438f81bf26979d9ea296331d265d/target/linux/ath79/dts/qca9531_glinet_gl-ar750.dts#L66-L67
```c
	i2c {
		compatible = "i2c-gpio";

		sda-gpios = <&gpio 17 GPIO_ACTIVE_LOW>;
		scl-gpios = <&gpio 16 GPIO_ACTIVE_LOW>;
	};
```
Instead of `GPIO_ACTIVE_LOW`, SDA and SCL should be `(GPIO_ACTIVE_HIGH | GPIO_OPEN_DRAIN)`.

Additionally, SDA is apparently on GPIO1 instead of GPIO17.

### Additional context
Upstream bug: https://github.com/openwrt/openwrt/issues/16319
Upstream PR to fix it: https://github.com/openwrt/openwrt/pull/16406

aanon4, 2024-09-17:
Is this a current problem? Otherwise I'l just waiting until OpenWRT rolls the change into a release.

ryanrsrs, 2024-09-17:
I think I can backport the fix so it'll be picked up in OpenWrt 23.05.5.

bwarden, 2024-09-17:
> Is this a current problem? Otherwise I'l just waiting until OpenWRT rolls the change into a release.

It's currently broken, but up to you whether it's easier to wait for an OpenWRT release with it included. I can open a PR with the patch added to `patches/` and `series`; I'm just waiting for CI to complete on the upstream PR.

aanon4, 2024-09-17:
Open the PR for it, but unless someone needs this fix today I will probably wait until after the next release to roll this in.

ryanrsrs, 2024-09-24:
OpenWrt 23.05.5 has been released and contains this fix.

closed 2024-10-14 by aanon4