aredn/aredn #1527: New UI : Color theme / descriptive icon change

mathisono, 2024-09-17:
Ive taken a stab a the color themes, for light/dark/Color Blind* and updated 2 icons to better depict. 

The color theme is really only for light and dark but adhere to an accessible color pallet. They give the "mesh status"/ "Cloud Supernode" / "Map" icon(s) distinct colors. Ive all so suggested The  icon highlight color be  changed to the same color as found in the ARDEN logo. 

[image]

-Chosen for a better pictographic diction of "node status".
[image]

-Chosen for a better pictographic depiction of "Local Mesh" the Heroicons library lack pictographic depiction of mesh or network.
[image]

Request
*1 Theme text "Color Blind" --> changed to Color Deficiency" as it more accurately describes the condition without implying a complete lack of color perception, which is not usually the case with color blindness

aanon4, 2024-09-17:
While I like the idea of adding a little color to the edge icon, at this stage it will need to be punted until a subsequent release.

As for the "Color Blind" name - I talked to a number of Color Blind people and this was the unanimous request for the theme name.

aanon4, 2024-12-04:
Adding color to the icons turns out to be tricky as these colors most also adapted to different themes. So unless someone submits a PR for this which works in all cases, it's not worth the investment I'm afraid.

closed 2024-12-04 by aanon4

mathisono, 2024-12-06:
Using a universal colors, pallet would be my preference. Adding color to the Icons is needed to direct/help direct user interaction with the new UI.  The pallet is the same for all sc

colors gradients are already in used. 

where and how are the different colors gradients defined? How are the themes defined? Im not proposing themes, my intent is for them to be universal to the ui.  point me in the right direction... do the icons have a style definition? 

3 things...
1. Change the icons default color
2. Keep the Hover over color 
3. Change node info icon color that's distinctively different from the Hover over color.