aredn/aredn #1540: New UI

mathisono, 2024-09-19:
Previous UI gives message info that explains why LQM has blocked an RF link. 
The NEW ui only displays ide with no explanation. if this intended, if so, the new UI need some added error reporting in the Neighborhood Node tab. 

[image]
[image]
[image]

suggestion: add another unique icon to depict if local LQM has blocked or if a link is idle. Then add error reporting to Neighborhood Node tab. 

<svg xmlns="http://www.w3.org/2000/svg" fill="none" viewBox="0 0 24 24" stroke-width="1.5" stroke="currentColor" class="size-6">
  <path stroke-linecap="round" stroke-linejoin="round" d="M14.857 17.082a23.848 23.848 0 0 0 5.454-1.31A8.967 8.967 0 0 1 18 9.75V9A6 6 0 0 0 6 9v.75a8.967 8.967 0 0 1-2.312 6.022c1.733.64 3.56 1.085 5.455 1.31m5.714 0a24.255 24.255 0 0 1-5.714 0m5.714 0a3 3 0 1 1-5.714 0M10.5 8.25h3l-3 4.5h3" />
</svg>

or 

<svg xmlns="http://www.w3.org/2000/svg" fill="none" viewBox="0 0 24 24" stroke-width="1.5" stroke="currentColor" class="size-6">
  <path stroke-linecap="round" stroke-linejoin="round" d="M3 3v1.5M3 21v-6m0 0 2.77-.693a9 9 0 0 1 6.208.682l.108.054a9 9 0 0 0 6.086.71l3.114-.732a48.524 48.524 0 0 1-.005-10.499l-3.11.732a9 9 0 0 1-6.085-.711l-.108-.054a9 9 0 0 0-6.208-.682L3 4.5M3 15V4.5" />
</svg>

aanon4, 2024-09-19:
It doesnt just disable idle, it also displays blocked, bad, poor, okay and excellent. However, the details of why the link is blocked is in the pop-up only.  Also the color is meaningful. The online docs could probably say a little more about this.

mathisono, 2024-09-19:
[image]
[image]

mathisono, 2024-09-19:
It's good that there is error reporting it's just if it's in a hover over
pop-up only it's hard to take a picture of and especially when you're
looking at all of the statistics and error rates it's nice to have that
right next to it

aanon4, 2024-12-04:
The full information is now available as a popup even when not logged in. I'm not going to add more information or icons to the non-popup screen.

closed 2024-12-04 by aanon4