aredn/aredn #1543: Errors during firmware updates

m0xzl, 2024-09-19:
### Describe the bug
Errors have been encountered during FW updates this morning.
This is a follow-on from issue #1521 
Similar problems encountered on two nodes:
* M0XZL-GL1: AR150 (from: 20240917-ef66a2a, to: 20240918-0a3d322). Error msg "ERROR: Firmware not compatible with this device."
* M0XZL-GL10: AR150 (from: 20240917-ef66a2a, to: 20240918-0a3d322). Error msg "ERROR: Unknown error"

### Expected behavior
FW update should complete without errors. If there is an error the message should be helpful in determining what action to take next.

### Screenshots
AR150 (M0XZL-GL1)
[image]
[image]

AR150 (M0XZL-GL10)
[image]
[image]

### Additional context
* All controls on the modal performed as expected.
* Repeats of the update process were successful in both cases.

m0xzl, 2024-09-19:
Sorry, I attempted to paste in the support data files but it doesn't seem to have worked. Trying again.
[supportdata-M0XZL-GL1-2024-9-19-7-35.tar.gz](https://github.com/user-attachments/files/17054765/supportdata-M0XZL-GL1-2024-9-19-7-35.tar.gz)

[supportdata-M0XZL-GL10-2024-9-19-6-37.tar.gz](https://github.com/user-attachments/files/17054766/supportdata-M0XZL-GL10-2024-9-19-6-37.tar.gz)

aanon4, 2024-09-19:
Timing wise, I think you need to have 20240918-0a3d322 installed to get the latest changes to the Upgrade Firmware fixes (they were committed on the 17th, so in the 18th's build). So the *next* updates would be the most interesting ones.

m0xzl, 2024-09-19:
Understood

aanon4, 2024-10-04:
Closing as there has been nothing added in two weeks.

closed 2024-10-04 by aanon4